libcsp/libcsp #278: clang-format broke #22

yashi, 2021-10-31:
#22 says that 
```c
char *correct = "This is correct";
char * bad = "This is not";
```
But ce4df1e6b2789efd2de changed to the bad.

yashi, 2021-10-31:
https://github.com/libcsp/libcsp/blob/e000dc57cf245224df9ea98edc5e38a4cb15c08c/doc/codestyle.md?plain=1#L12

Ah, it's intentional.

closed 2021-10-31 by yashi

johandc, 2021-10-31:
Yes I have always used to put spaces on both sides, I think it looks prettier. With clang it was easy to rectify, so I went with the path of least change, which was to still keep it on my personal preference.

There seems to be no disadvantage on one over the other. Google prefers the pointer to the left next to the type. Jeppe preferred it to the right next to the variable name.

Maybe I'm wrong for putting it on the middle, maybe there os one way that is the most correct. I think there could be some cases with the usage of const where the asterisks could help show which part it belongs to.