libcsp/libcsp #294: Remove csp_sleep

johandc, 2021-11-13:
We shouldn't be wrapping functions which are POSIX standard

Prefer to use unistd.h and usleep, or better
time.h and nanosleep which are POSIX 2008.

yashi, 2021-11-14:
I'm fine removing `csp_sleep()`.  But no sure about relaying on POSIX.

Because we are using `gnu11` as `c_std`, we don't have to worry about POSIX.  But there are many OSes and development environment without POSIX, let alone with POSIX 2008 or up coming new POSIX standard.   picolibc ls pretty great C library.  But I don't know it has full POSIX support, either.

If we want to support non-GNU or non POSIX OSes, I'd suggest switching `c_std=c11` for libcsp.  So that we can be C11 clean.

If we remove `csp_sleep()` we can get rid of the usage from `csp_if_udp.c` and `can_socketcan.c`.  These are something I've wantend to fix.  Having `sleep()` doesn't indicate a good practice, IMHO.

johandc, 2021-11-14:
Well one way we could effectively get rid of it, was if we use csp_task_wait() instead. This uses a task notification timer to wait, which is better than calling sleep(). For barebones, this results in a sleep() and fore RTOS' and friends, it results in a Task being yielded and waiting for that amount of time. 

`csp_if_udp.c` here it's used when the CSP stack is out of buffers in order to avoid using 100% CPU in the RX loop. I believe the correct thing to do in this situation is in fact to busy wait, and allow for the consumer to free up more buffers. Since we are on POSIX for if_udp already, we could safely just call nanosleep directly.

`can_socketcan.c` the use of sleep here is to throttle the TX when the linux CAN driver is out of MOB's. This typically only happens if there is a break in the CAN wires. It is retrying the TX every 5 ms up to 1000 ms before giving up. I think this is an acceptable use of the sleep call. On POSIX sleep() is not implemented using a busy-loop, so it's pretty efficient way of blocking and retrying.

Both the above mentioned uses of csp_sleep_ms, could be using the csp_task_wait() function as suggested. Because it would reduce the number of hooks the user has to implement.

johandc, 2021-11-16:
I decided to go with the older <unistd.h> and sleep / usleep. I checked all toolchains i could find, avr8, avr32, picolibc and all newlib based ones have <unistd.h>

commit 55e9142a (2021-11-16): Got rid of csp_sleep.h

Fix #294
closed 2021-11-16 by johandc