libcsp/libcsp #295: Remove queue and semaphore

johandc, 2021-11-13:
Okay here is a biggie, but hear me out.

I believe it is possible to get rid of mutex, by using atomics. Generally this has been easy to figure out. And it saves a Lot of system call work, and works with hardware instructions on most modern platforms. 
(We do still use it one more place which is on the uart driver i believe. )

As for semaphore, we have one use in zmqhub where it's used as a tread guard, because libzmq isn't threadsafe. Here multiple threads are contesting one object. Which is a nice case for a mutex actually. However since zmqhub is Linux only, I feel like this should be solved better somehow. Maybe a Linux specific method.

The final place where semaphore is used in rdp to take the tx task when more work needs to be done. For this where only one task is to be woken. We should be able to do it with a simple wake callback on the conn object. Similarly we need a sleep callback I'm order to suspend the tx task. This is simple and avoids a lot. And of the user wants he can simply ignore sleeping and waking and use polling instead.

I believe we can use something similar instead of the RX queue object for the router task to wait on. By replacing the synchronization of the router task and the input foto, with the aforementioned sleep and wake callbacks we can pretty much get rid of queues too. 

Of course CSP would still rely on queueing. But it's not os dependant any longer. And can be made using simple ringbuffers. 

This is a lot to take on. But i believe it would work. So CSP could actually operate completely without an OS. Neat huh.

yashi, 2021-11-14:
### locations used except examples:

**semaphore**
- `transport/csp_rdp.c`
- `interfaces/csp_if_zmqhub.c`

**mutex**
- `interfaces/csp_if_kiss.c`

**rdp**
Even with a callback, we need to serialize the callbacks.  How do you wait from the calling thread for a callback to be finished?

**zmqhub.c**
we can just use pthread semaphore.  But the usage here in zmqhub.c is actually mutex.  so, i'd go with pthread mutex.

**if_kiss**
I don't know what the driver is trying to protect.

### thoughts
1. This led me some questions.  Is libcsp thread safe? Or intend to be, at least?  we have some shared global variables, ie `rtable_inptr`, `interfaces`, but we have no locks around it.   We've never noticed any errors because most of modifying functions, `csp_rtable_set_internal()`, `csp_iflist_add()`, runs at startup time and no one has called in the middle of the operations.  Nor, it's just rare case to call these functions.
2. libcsp doesn't actually differentiate mutex and binary semaphore.  But many OSes do.
    
    https://www.freertos.org/xSemaphoreCreateBinaryStatic.html
    > Binary semaphores and mutexes are very similar but have some subtle differences: Mutexes include a priority inheritance mechanism, binary semaphores do not. This makes binary semaphores the better choice for implementing synchronisation (between tasks or between tasks and an interrupt), and mutexes the better choice for implementing simple mutual exclusion.
    
    But many OSes has a property of owner on mutex.  Recursiveness is another property of mutex.
    
    https://www.freertos.org/xSemaphoreGiveRecursive.html
    > A mutex used recursively can be 'taken' repeatedly by the owner. The mutex doesn't become available again until the owner has called xSemaphoreGiveRecursive() for each successful 'take' request.
    
    https://docs.zephyrproject.org/latest/reference/kernel/synchronization/mutexes.html#concepts
    > An owning thread that identifies the thread that has locked the mutex, when it is locked.
    
    https://docs.windriver.com/bundle/vxworks_7_application_core_os_sr0630-enus/page/CORE/semLib.html
    > For a mutex, a semGive( ) can only be performed if the current task is the owner, implying that the semaphore has been taken; thus, there is always a change of state.
    
    https://www.keil.com/pack/doc/cmsis/RTOS/html/group__CMSIS__RTOS__MutexMgmt.html
    > The advantage of a mutex is that it introduces thread ownership. When a thread acquires a mutex and becomes its owner, subsequent mutex acquires from that thread will succeed immediately without any latency. Thus, mutex acquires/releases can be nested.
    
    https://pubs.opengroup.org/onlinepubs/7908799/xsh/pthread_mutex_lock.html
    > This operation returns with the mutex object referenced by mutex in the locked state with the calling thread as its owner.

johandc, 2021-11-14:
Yes, you see what a mess that is...

I would like to address the three examples in reverse order:

3) KISS - The Mutex is there to ensure only a single task is transmitting to the UART driver at a time. And therefore only full frames are transmitted. Our UART driver API actually sends out bytes one at a time, so even though the underlying UART driver would ensure string-level consistency, CSP doesn't submit as whole strings (partly due to a memory issue with that). So we actually do need the mutex.

2) ZMQHUB - I agree, first of all, it should have been a mutex. And secondly, we should just use the native API on that platform. Maybe we should even check again, if the ZMQlib thread safety has improved since then. Because i would expect something like ZMQ to be just that. Considering that it's an inter-process communication tool in the first place. One would assume thread safety was a priority :)

1) RDP. This was the hardest one, because it actually uses the semaphore for real task sleep, and wake event. So in order to "fix" that, without being OS dependent, i would first like to elevate the issue to the more general one:

**Question: How to make a caller block, on an event coming from the CSP stack?** 

The general case is used multiple places. csp_listen(), csp_read(), csp_send() - if RDP is enabled. 

**Answer 1: The NON solution**
The NO-OS method, would simply be to return from the function call with an error and let the user handle getting back. IE. to NOT support blocking. I don't like this, because it changes the API. We should keep the blocking variants, and add special non blocking variants, designed for NO-OS.

**Answer 2: Add csp_task_wait() and csp_task_wake() hooks, that the OS can implement**
Whenever, the CSP stack encounters a place where blocking is required, csp calls csp_task_sleep which puts the current task to sleep for a specific amount of time (as requested by the user in the listen, read, send or queue call). When the sleep is completed by timeout, the function returns to the code that called the sleep, and resumes processing. If an interrupt, or another task causes the bloking item (a connection queue, for csp_listen, a conn_rx queue for csp_read() or an ACK packet is received and more TX is requested by RDP) to be woken, a call to csp_task_wake() is performed. This holds the reference to the task. CSP will keep the task handle associated with the connection, so it knows who to wake.

**Question: How would sleep and wake be implemented?**

**Answer: FreeRTOS**
We use the very nice and lightweight [Task Notification](https://www.freertos.org/RTOS-task-notifications.html) function: See example [here](https://www.freertos.org/RTOS_Task_Notification_As_Binary_Semaphore.html)

**Anwer: Posix**
I asume we have something similar to freertos?

**Anser: Baremetal**
Baremetal should not use blocking, since it would cause the executing task to actually block the CPU. However, IF someone would simply want to wait a very short time, it could be done using an actual microsleep() call, or any other delay function. One other way would be in the case of Co-Routines, to make a call to taskYield(). (using freertos terminology here)
Note: listen() and read() is easy to pass a zero timeout to obtain non-blocking. But csp_send() actually recently got the timeout and return value removed. And if RDP is enabled on a connection, the csp_rdp_opt of the client, determines what the packet timeout and connection timeout is. So the answer is, the caller to csp_send actually don't know if it would be blocking or not. So for RDP i suggest we add csp_send_nonblocking() which is capable of returning false, if the packet was not accepted. csp_send() is fire and forget. And is guaranteed to always deliver or drop the packet. So a new csp_send_nonblocking() must be the answer here.

**List of benefits**

1) Extremely lightweight, almost zero memory impact
2) Zero OS dependencies
3) User can port to his OS of choice by implementing hooks a concept we already use for time, memfree, reboot etc.

**List of disadvantages**
1) wait() and wake() can only block a single task on a single connection. - This is not an issue with CSP

yashi, 2021-11-15:
**KISS**
> to ensure only a single task is transmitting to the UART driver at a time

I don't know why we need this restriction.  If accessing a device from multiple threads is the problem, we can push the lock down to the drivers.  is there any issue at the interface between IF driver and UART driver?

**ZMQHUB**
Have we had any problem?  4e7d4fa490e9805b introduced the lock but nothing is mentioned.

**Notification API**
Maybe because I don't know much about FreeRTOS, I don't see the benefit of introducing task notification API.  What API do you have in your mind for POSIX?

I don't want to be negative about it but:
- I don't know how this can be "Zero OS dependencies", you have to port to each target platform.
- "Extremely lightweight" is implementation detail.  It could be heavier than Mutex for some OSes.
- If we have less than a hand-full of places for mutex and semaphore, do we care the memory reduction?  How many bytes can we save?
- It's not widely understood API.  Developers already know mutex and semaphore.

johandc, 2021-11-15:
KISS.
If we push the lock down to the driver, we would need to prepare the kiss packet into a temporary buffer, so the data to transfer could be done within a single call to the driver. I'm not sure that is practical, due to the memory allocation issue. And unfortunately kiss inline byte stuffing means the encoding cannot happen inplace. But we could make a new lock API just for the driver itself. So CSP would first ask the driver to retrieve a mutex, and then free the mutex again. Something like csp_uart_lock() / _unlock() would work fine. And move the os dependency into the driver layer. Where we already arch specific code in the driver folder. Which should work nicely.

ZMQ.
The documentation states that the zmq context object is thread safe. But the individual sockets are not thread safe. The idea is that each task using ZMQ should hold its own set of sockets for transmissions. This is impractical for CSP, since we cannot hold any task specific information. So a mutex seems to be the appropriate solution. I don't know why a semaphore was chosen and why it was called tx_wait. I believe that actually it's just a mistake. Because it could have been copied from RDP send where the tx semaphore is called tx_wait. I believe for zmq we should just use a POSIX mutex.

So with this. We have now removed the need for a generic CSP mutex API. At the expense of only one additional set of lock/unlock calls to the uart driver layer. I think that's nice.

I consider the issue solved for mutex. I'm addressing the task synchronization in the next comment which may be tomorrow. As it's late.

johandc, 2021-11-15:
> 
> **Notification API** Maybe because I don't know much about FreeRTOS, I don't see the benefit of introducing task notification API. What API do you have in your mind for POSIX?
> 

`pthread_cond_wait()`

> I don't want to be negative about it but:
> 
> * I don't know how this can be "Zero OS dependencies", you have to port to each target platform.
> * "Extremely lightweight" is implementation detail.  It could be heavier than Mutex for some OSes.

Perhaps. But for FreeRTOS and POSIX it would work out nicely. I think a binary semaphore can be considered a notification. But it comes with the semantics of being a semaphore. So one would usually say there needs to be as make entering and leaving. With the way we use it, for synchronization, we might as well use an API only designed for that. The notification system is designed to be just that. Which means usually more lightweight. And you don't care if anybody is listening. You just send the notification whenever an event happens. And if someone was waiting they will be unblocked.

For FreeRTOS this is additionally extremely lightweight as each task has its own method for unblocking built-in. So there is zero memory required.

> * If we have less than a hand-full of places for mutex and semaphore, do we care the memory reduction?  How many bytes can we save?

Oh, we have many. For each connection there is one for RX. One for RDP RX, one for RDP TX. So on a typical MCU with 10 connections this amounts to 30 semaphores. A semaphore in freeetos is a queues with length 1. So the entire queue structure follows. Which is around 32 bytes each. So it amounts to at least 1kb of RAM just for synchronization semaphores within CSP. Which isn't needed at all.

On our subsystems we use the largest cortex m0 possible, which holds a whopping 32 kB of RAM only. Now you see why 1 kB additional memory would be nice to have.

Also I don't like to waste ressources where I can see a prettier solutions.

> * It's not widely understood API.  Developers already know mutex and semaphore.

They also know conditional wait pretty well. And task synchronization in general. Task notify has been available within FreeRTOS for around 5 years only. So it's quite new. But we would of course include the implementation for these hooks in CSP so the users of CSP wouldn't even know.

yashi, 2021-11-15:
To use POSIX cond var, you have to have mutex:
```c
int pthread_cond_wait(pthread_cond_t *restrict cond, pthread_mutex_t *restrict mutex);
```

Zephyr and other RTOS' condition variable also requires to have a mutex.  Because most implementations of condition variables are actually not only a condition variable.  You use a condition variable _and_ a mutex to have a Monitor. It's basically creating a block of mutual exclusion for accessing the resource protected by the mutex, but can atomically wait on the it.

Can we just implement binary semaphore using the notification API for FreeRTOS?  So that FreeRTOS can benefit everything you described.  But other OSes can keep using semaphore API as they used to.  It certainly is NOT  "Zero OS dependencies".  All OSes except FreeRTOS have to implement this new API.

I've read a bit about FreeRTOS task notification, and it seems like a Unix signal.  You wait for a number, you'll be woken by the number.

johandc, 2021-11-15:
> To use POSIX cond var, you have to have mutex:

Okay, weird. I read it like, if a Mutex was passed it will automatically release the lock and unblock the correct task. Assuming that if no mutex was passed it would simply wake the task.

So you are right in that what I am looking for, is a simple wake signal () however signals in Unix are a mess, and can only be used on processes. Threads don't work with signals.

I can see that the method pthreads librt uses to implement semaphores, is in fact a conditional variable and a Mutex. So on Linux at least, you always get that additional overhead. So it doesn't matter of we use semaphore.h to get the sem_wait() function or we use cond_wait with our own mutex.

I think it is perfectly fine to do notifications for FreeRTOS and binary semaphore for others. But instead of using the arch layer, and preprocessing macros to change the data structures to hold posix_sem_t * for example, I would rather implement the more general wait() and wake() API.

On posix and zephyr this can use an ordinary semaphore, on FreeRTOS we use lightweight notifications.

I wanted to make it so, after the call to wait () the task will input a generic handle, void *, to be used by the wake function. Which leads the question on where to store the semaphores.

On FreeRTOS there is no need for storage, so the wait() Will simply store the task handle it wants to wake up directly.

On POSIX the semaphores is created dynamically, so here we store the semaphore handle to wake on.

On zephyr I guess it's a little harder, because you need a larger memory block to hold the storage for the semaphore object. But I assume you want to avoid dynamic allocation. I think the solution is to create a generic reservation on the conn element, which can be compile time configured. That way the os hooks can put whatever information into the conn object it wants.
This approach is similar to how drivers borrow room in the padding area of the CSP buffers. And is really efficient. I like this idea.

johandc, 2021-11-16:
**Status**
- `transport/csp_rdp.c`
- ~~`interfaces/csp_if_zmqhub.c`~~ Uses native pthread mutex
- ~~`interfaces/csp_if_kiss.c`~~ Moved to UART driver

johandc, 2021-11-16:
I have now vastly cleaned up the bin_semaphore api, so csp_rdp.c can continue to use this for now. I tested it on posix, and it builds on freertos. I will continue to move freertos to the even simpler/smaller task notify api later on.

Eventually i am going to take a look at queues too. I would like to add a no-os implementation of queues. I think we will leave the freertos and zephyr queues alone for now. It's going to actually grow CSP to use it's own queue implementation.

johandc, 2021-11-18:
Okay today I tried out and added the task notification API of FreeRTOS. Also POSIX os completely static allocation for binary semaphore.

I did as you suggested and put the FreeRTOS notifications behind CSP binary semaphores.

On FreeRTOS a real binary semaphore which is based on a queue with depth = 0, takes up 76 bytes. Where the new API works just as well and takes only 4 bytes.

Right now just doing this in rdp saved me 72 * 20 connections = 1.4 kB of memory on a small microprocessor with just 32 kB RAM.

So i think thats significant.

I will close this for now, you will likely have to fix the zephyr semaphore.c with a cast.

Maybe I will revisit queues later to make a no-os queue implementation too.

closed 2021-11-18 by johandc