libcsp/libcsp #297: Extending address space beyond 32 addresses

jgromes, 2021-11-15:
Hi,

first of all, thanks for making this open-source!

I've been working with CSP for a while now, and some time ago, hit a roadblock when designing a CSP network with more than 32 nodes. I do understand the address range was a design choice, and is probably the best compromise for a 4-byte header. However, I was wondering - is there a "recommended" way to extend CSP to more than 32 nodes?

My solution to the issue was basically a NAT (network address translation) router, separating the CSP network into sub-networks and performing packet packing when routing traffic between subnets. However, I find this solution a bit ugly. It leads to things like multiple instances of CSP running on a single node, NATing tables alongside routing tables or having to deal with out-of-sequence connections.

All ideas are very welcome!

johandc, 2021-11-15:
Funny that you ask.

Check out the developmenr branch changelog.
CSP 2.0 is just around the corner with 16k addresses

closed 2021-11-15 by johandc

jgromes, 2021-11-15:
> 16k addresses

Cool! Though I'm probably going to stick with my NATing solution (despite it being absurdly overengineered), seeing that I need it to be compatible with CSP 1.0 :P

> just around the corner

That's great! When can we expect an official release?

johandc, 2021-11-15:
We are still hard at work at the moment making as many breaking changes while we are at it. So hopefully CSP 2.0 API will remain stable for almost as long as CSP 1.0 :)

We have been using CSP 2.0 for several satellites already in orbit, so it's just a matter of making the decision to move. Products that run on 2.0 can be configured at start-up to run 1.0 instead.