libcsp/libcsp #306: latest build requires meson 0.56 which is not available with ubuntu-latest on github actions

johandc, 2021-12-02:
Github actions run on 'ubuntu-latest' which is pretty old 20.04. We can either wait a few months for 22.04 to become available on github. Or we can spin up a docker of ubuntu:21.04, or we could run an install script that uses pip install meson, because that is always newer than the 'apt install meson' version.

yashi, 2021-12-03:
What feature do you need from 0.56? Using docker just for meson is overkill.  We can get the latest from some other channel, ie. PyPi.

johandc, 2021-12-03:
we need the dependency.as_link_whole()

Can you update the script so it uses pip install to get meson instead?

yashi, 2021-12-04:
which function are you in your mind?
Does #308 help?

yashi, 2021-12-04:
Or use `both_librarries()` and let Meson choose the right lib.

johandc, 2021-12-05:
It is specific to the usage of weak linked functions from,  inside of a library. The only way that can work when using static linking is by using the link-whole statement. Dynamic linking is not the solution.

I am in doubt if the weak function are the correct way in general. See hooks.md in the docs folder for an overview of the hooks we currently have. Two of them, the print and the clock get have default implementations within the library.

The callbacks for system reboot example is also the wrong method.

The final method is using preprocessor macros, which can quickly get messy when there are a lot of overrides.

johandc, 2021-12-05:
Actually there is another method that I am considering.

And that is to remove the weak part of the linking, and always require the user to implement the hooks. And then have default implementations within contrib for various platforms.

A further optimization of this is to have macros like HAVE_INPUT_HOOK = y to remove calls to non implemented hooks.

This is the approach that FreeRTOS uses btw.

johandc, 2021-12-05:
Just as described here.
https://www.freertos.org/a00016.html

yashi, 2021-12-05:
Please don't.  Why do we have to go back to '80s when our compiler and the elf format provide better and smart way to do?

johandc, 2021-12-05:
I went for the weak linked functions :)

And removed the weak from the declarations and added the always defined weak functions for all hooks.

closed 2021-12-05 by johandc