libcsp/libcsp #308: Remove weak attribute from function declarations

yashi, 2021-12-04:
Weak attributes are attributes for functions; Not for function declarations.  We have `__attribute__((weak))` in some header files and they must be removed.

johandc, 2021-12-05:
This is not true.

If the functions are not declared as weak, it is impossible to test for their implementation like this

```c
__attribute__((weak)) void csp_reboot_hook(void);
 
if (csp_reboot_hook) {
    csp_reboot_hook();
}

```

Removing the weak from the declaration produces this warning:

```
../lib/csp/src/csp_service_handler.c:300:37: error: the address of ‘csp_reboot_hook’ will always evaluate as ‘true’ [-Werror=address]
  300 |                                 if (csp_reboot_hook) {
```

yashi, 2021-12-05:
What is the point of checking the function pointer?  We _always_ supply the weak version of the function as default.  We use weak attribute so that user will _override_ the default function we supply.  Otherwise, we don't even have to use weak attribute.  Have a function pointer and let user set it.

yashi, 2021-12-05:
**main.c**
```c
void csp_reboot_hook(void);
int main()
{
	csp_reboot_hook();
	return 0;
}
```
**weak.c**
```c
#include <stdio.h>
__attribute__((weak)) void csp_reboot_hook(void)
{
	printf("weak\n");
}
```
**strong.c**
```c
#include <stdio.h>
void csp_reboot_hook(void)
{
	printf("strong\n");
}
```
**build**
```shell
gcc -Wall -Wextra -g main.c weak.c && ./a.out 
weak
gcc -Wall -Wextra -g main.c weak.c strong.c && ./a.out
strong
```

johandc, 2021-12-05:
Okay, i see your point; That if we always provide a weak implementation and if people remember to use --link-whole then we do not need to check. One of the reasons the CI was crashing (segfaulting) was that i called a weak function (csp_print) when the program was linked with libcsp.a without --link-whole. Which `waf` does not support. 

I have added default implementations to all hook functions. Even just empty ones. So we do not need to perform the check. We should keep an eye out for crashes now.

closed 2021-12-05 by johandc