libcsp/libcsp #312: The big 2.0 cleanout: Removing old features

johandc, 2021-12-07:
If you haven't noticed, it started already.

So far i have gotten away with slating a few old things related to the insides of CSP.

@bfol Raised a concern about TaskList which was a bit too far because it affects the actual on-the-wire protocol. This is why we need to start a discussion about that. So far i want to talk about the following:

- Not related to networking:
  - [ ] `TaskList`
  - [ ] `memfree`

- Unmaintained code
  - [ ] `XTEA` - I cannot maintain this. It's basically frozen unmaintained crypto code which is never a good thing.
  - [ ] `HMAC` - Pretty useful, but security and access control should be handled outside of CSP in my oppinion. At the network boundaries.

- Harmfull code
  - [ ] `SFP` relies on malloc so the implementation is broken at least. Because there is no retransmission and all data is moved by copy and everything needs to arrive in the correct order. I think it is more harmfull than good.

Q: **Do you agree on the removal of these?**

Q: **If yes, when is it okay, to remove any of these features?**

johandc, 2021-12-07:
> Q: **Do you agree on the removal of these?**
> 

Well, yes... I suggested them.

> Q: **If yes, when is it okay, to remove any of these features?**

I Think SFP, XTEA, and HMAC should go for the major 2.0 release. I'm 95% sure that no-one actually uses them. 

`TaskList` and `memfree` i think we should support with a user defined hook for the foreseeable future. Maybe 5 years or so, until all users have turned away from using them.

bfol, 2021-12-08:
Q: Do you agree on the removal of these?

From gomspace side, we use TaskList and memfree. We would hate to see these go, but are ok to move the server side implementation to a hook, so the arch support can be removed. 
csp_hex_dump is not used by us at all.
We also use HMAC and SFP, and object to removal of these. XTEA is not used by GomSpace. 

Q: If yes, when is it okay, to remove any of these features?
My general view, is that we should be very careful to remove anything not being deprecated for for a few major releases. 

On a side note, I will really appreciate if we can slow down on the cleanup and changes being introduced all the time, so that we can focus on getting the I2C support ready and test / stabilize the code for the upcoming 2.0 release. Otherwise we are trying to hit a moving target.

johandc, 2021-12-08:
Fair points all of them.

So how about we for 2.0:
  Keep but deprecate SFP, or at least give notice about it's potential harmfulness. 
  Keep HMAC untouched
  Keep TaskList and memfree as OS dependent hooks
  Remove XTEA
  Remove csp_hex_dump

I agree to slow down for 2.0 and make this the end of changes.

bfol, 2021-12-09:
I would prefer not to deprecate SFP, but add a note on the potential harmfulness, otherwise I agree to the the proposals. 
I will then start to work on the I2C interface changes.

closed 2021-12-27 by johandc