libcsp/libcsp #318: sem: Why do we need alignment hack for all arches

yashi, 2021-12-18:
https://github.com/libcsp/libcsp/blob/de88960c9502c80931cef5272d13b49921586a8e/include/csp/arch/csp_semaphore.h#L48-L57

@johandc, would you mind elaborating me why we needed this alignment thing for _all_ arch?

johandc, 2021-12-18:
Yes. I thought it would be nice for all semaphores to be statically allocated. This was previously done with a macro that resolved to some arch native type. So a csp_sem_t was not an actual type, it was an OS thing that put a requirement on an external header. Like FreeRTOS.h or zephyr.h for anybody who uses the semaphores. So i guess it helps reduce the number of headers that gets pulled in.

I don't know which is better. To use the union to perform the alignment or the GCC attribute align.

yashi, 2021-12-19:
> I thought it would be nice for all semaphores to be statically allocated.

I agree.

> So i guess it helps reduce the number of headers that gets pulled in.

Why did you want to reduce the number of header files to be included?

Header files are there to be used.  Let's use it when we need it.

I can keep this tricked typdef for FreeRTOS like the following or convert to simple `typedef StaticSemaphore_t csp_bin_sem_t;`.  What do you think?

```diff
diff --git a/include/csp/arch/csp_semaphore.h b/include/csp/arch/csp_semaphore.h
index 4b47616c..e605ecd3 100644
--- a/include/csp/arch/csp_semaphore.h
+++ b/include/csp/arch/csp_semaphore.h
@@ -22,9 +22,8 @@
  */
 
 #if (CSP_POSIX || __DOXYGEN__)
-
-    /* POSIX interface */
-    #define CSP_SIZEOF_SEM_T 32
+#include <semaphore.h>
+typedef sem_t csp_bin_sem_t;
 
 #elif (CSP_FREERTOS)
 
@@ -35,16 +34,6 @@
 #define CSP_SIZEOF_SEM_T sizeof(StaticSemaphore_t)
 #else
 #define CSP_SIZEOF_SEM_T sizeof(void *)
-#endif
-
-#elif (CSP_ZEPHYR)
-
-    /* Zephyr RTOS Interface */
-    #include <zephyr.h>
-    #define CSP_SIZEOF_SEM_T sizeof(struct k_sem);
-
-#endif
-
 /**
  * This definition is borrowed from POSIX sem_t
  * It ensures the proper amount of memory to hold a static semaphore
@@ -56,6 +45,13 @@ typedef union {
 	long int __align;
 } csp_bin_sem_t;
 
+#endif
+
+#elif (CSP_ZEPHYR)
+#include <zephyr.h>
+typedef struct k_sem csp_bin_sem_t;
+#endif
+
 /**
  * initialize a binary semaphore with static storage
  * The semaphore is created in state \a unlocked (value 1).
```

johandc, 2021-12-19:
I don't want any implementation specific headers in the public CSP headers. The reason is that CSP should not rely on external headers compile time, for people who are using our API.

 However one could argue that since we discourage the use of #include csp/arch in the first place, we might as well just move the header file to internal path. And then I agree that using the typedef is prettier.

yashi, 2021-12-19:
> I don't want any implementation specific headers in the public CSP headers.

We have inconsistency.  csp_queue.h has arch specific things.

> The reason is that CSP should not rely on external headers compile time, for people who are using our API.

Why?  We can't basically avoid that.  If you are using FreeRTOS and CSP needs some kernel specific features, we must build against it.  That's because we must know the size of the object at the compile time.  And to avoid it, the trick you used is to put a literal number for the size of object by assuming the size won't change?

It's always nice to reduce dependencies. But, when we need it, we must use it.

> just move the header file to internal path

How about:

- `include/csp/csp_semaphore.h` for the public header
- `include/csp/arch/csp_semaphore_freertos.h` for arch specific internal header.

I don't want to break the current applications at this stage so I'd
- Keep `include/csp/arch/csp_semaphore.h` for 2.0
- Create `include/csp/csp_semaphore.h` for the new header file which includes `include/csp/arch/csp_semaphore.h`
- After 2.0 is released,
  - Warn when `include/csp/arch/csp_semaphore.h` is included directly, and
  - Tell users the old path is deprecated and must use `#include <csp/csp_semaphore.h>`

yashi, 2021-12-19:
I've got an idea.  This is based on one rule.

> OS agnostic source code does NOT allocate OS specific object.

It's simple and acceptable, right?

Pros are
- We don't need to know anything about OS specific objects in our public header
- OS specific code must access the object, thus needs OS specific header file
- Application is OS specific anyway and must include OS specific header file

Cons are
- Our header files are not self-contained.  Some coding style requires it ([google](https://google.github.io/styleguide/cppguide.html#Self_contained_Headers))

Here is sample code

**application code**: `main.c`
```c
#include <stdio.h>
#include <someos.h>
#include "lib.h"

int main()
{
	Foo f;
	func(&f);
	printf("%d\n", f.a);
	return 0;
}
```

**public header**: `lib.h`
```c
struct foo;

typedef struct foo Foo;

void func(Foo *f);
```

**OS specific library source**: `lib_someos.c`
```c
#include <someos.h>
#include "lib.h"

void func(Foo *f)
{
	f->a = 100;
}
```

**OS header file**: `someos.h`
```c
struct foo {
	int a;
};
```

**How to build**
```shell
gcc -Wall -Wextra -g -I. main.c lib_someos.c && ./a.out
```

johandc, 2021-12-19:
That is known as an opaque pointer.

It is frequently used to hide the implementation of something. As for example the task control block (tcb) of FreeRTOS is an opaque pointer.

I like this approach because it also eliminates a large if/elseif block of code that needs to be maintained. Making the size available to the allocating unit, through a someos.h header seems very reasonable.

Sometimes only parts of the data structure is made available, and the rest is undocumented or obfuscated by variable names such as hidden, or reserved. While on the inside there is full access. The whole idea is to make the user responsible for the buffer, but to disallow any access or knowledge about the contents.

So I applause that as a solution in general.

However. To answer your question from above, I guess the reason I don't want FreeRTOS headers in the public API is still valid, because, I don't believe that csp_sem should be part of the API at all. There are several other os abstraction layers out there. So CSP should only contain what it needs to operate itself. Calling as few hooks and requiring as little from the OS as possible. The ultimate goal is still to be able to run on barebone too.

This hack is literally directly out of the posix semaphore implementation. It was much more beautiful than the rather complex FreeRTOS static semaphore setup which required a handle, a queue, and a queue item buffer. So i rewrote the semaphore from consisting of three objects, and a complex init function, to just a single object which is what's nice about that posix implementation. I can see that zephyr also uses the same notion with a single object. 

However now that I got the much much more efficient vTaskNotify to work on FreeRTOS. The static semaphore is in fact just a void pointer to the task to wake up. So it got even better.

In conclusion, we should work on getting the queue and the semaphore out of the API.

yashi, 2021-12-19:
> I don't believe that csp_sem should be part of the API at all.

OK.  I think I now understand that you only want use `csp_sem` _internally_, because libcsp proper needs some locking mechanism, but not wanting to expose the internal semaphore to the applications. Applications, OTOH, can just use whatever they have.

Let me work on that.

johandc, 2021-12-19:
Yes exactly. The same with queue. 

CSP does need that. But I was wondering if it was prettier to write our own input queue management using just <stdatomic.h> and a ringbuffer. Together with the csp_sem for task notification that should be all that is needed.

If you look at the amount of work done in FreeRTOS for queues and similarly on POSIX. It can all be simplified by the fact that:

Any queue or semaphore in CSP only ever has one consumer and one producer. So queues can be reduced to an atomic update and a means for synchronization. 

This latter could be a POSIX or zephyr semaphore, a FreeRTOS task notification or even just a barebone os setting a flag somewhere outside the library.

I think this is what is really necessary for CSP to run on baremetal. Which would be cool. Because so far. Half of CSP development has been around porting to different environments.

yashi, 2021-12-19:
> Because so far. Half of CSP development has been around porting to different environments.

Right. I'd love to get done with porting and work on libcsp proper. :smiley:

I don't know I can do as I say but I'll move `csp_sem` to internal, add configuration option to expose it as public API (default expose) in order to keep it compatible for v2.0.  If you beat me to it with the new implementation, I'll drop my PR.

How does that sound?

johandc, 2021-12-19:
I think we don't need to make an option to export the header. People moving to CSP 2 should change their code. CSP 2 is backwards compatible (in legacy mode) on the interface only. Not the full API. So I think it's as simple as moving the header file and modifying the typedefs.

PR #328 (2021-12-26): csp_semaphore: Use native semaphore type
All arch have a native semaphore type.  csp_bin_sem_t is now an internal type and we can pull platform specific headers in it.

For FreeRTOS, we are keeping the sizeof hack for this commit.

This closes #318.
commit 5ee7b895 (2021-12-26): csp_semaphore: Use native semaphore type

All arch have a native semaphore type.  csp_bin_sem_t is now an
internal type and we can pull platform specific headers in it.

For FreeRTOS, we are keeping the sizeof hack for this commit.

This closes #318.
commit 6d1d57b6 (2021-12-26): Remove CSP_SIZEOF_SEM_T for FreeRTOS

Fixes #318
closed 2021-12-26 by johandc