libcsp/libcsp #319: thoughts on csp_print()

yashi, 2021-12-19:
`csp_print()` is a macro but `csp_print_func()` is a function.  `csp_print()` doesn't pass any file related information to `csp_print_func()`. 
 That means that we can't get any file related info, such as `__LINE__` or `__file__` in `csp_print_func()`.

There is no way to replace `csp_print()`.

Also, some OS can print without stdio.  ie) having simple UART driver without full `stdio.h`.  But currently we are either
- having `stdio.h` and `csp_print_func()`, which uses `stdio.h`, or
- `csp_print()` become noop.  no print.

We need third option.  Enable `csp_print()` without `stdio.h`.

To combine the two, I propose: Make csp_print() replaceable so that custom csp_print() can take context info if needed.

johandc, 2021-12-19:
Something just like 

'''c
#ifndef csp_print()
#define csp_print() my_printf()
#endif
'''

That was my original proposal. But I went with a hook and weak linking after having that rather long talk. 

Remind me again what's the advantage of the hook? I guess the fact that the user does not have to specify either command line arguments, or inject header files into the include path of the CSP build is one. I guess that maybe I believed a linked hook was better. Maybe it is.

What you are proposing is for the hook to be called without the need for stdio. So maybe just put ind another compilation option to select stdio or print independently, and then remain with the hook implementation?

yashi, 2021-12-19:
Hooks are fine as long as no dependency is required.  The current `csp_print()` _requires_ to have stdio.  Without stdio, you can't even override `csp_print()` because you failed to build the weak version of `csp_print()`.

The complication is that with `csp_print_func()`, you can't get `__LINE__` or `__file__`.  This is special for debugging print function.  No hooks usually want to get calling site information.

> another compilation option to select stdio or print independently

Yes.  If you agree, I'll work on a PR along the line.

johandc, 2021-12-19:
Finally regarding the context info. The use of __FILE__ is considered bad for build reproduction. It literally changes the object size based on what folder you are compiling from. This can be a huge killer if something works when compiled on bo's machine but fails on peter's just because his home folder path is longer. I really dont see the reason for this, since all printf are rather unique and very easy to grep the source code for.

And finally let's get rid of CSP print. Let's look at each individual need for producing an output and see what can be done to prevent that from coming from inside a library.

Outside the library in examples we should just use the native print method of the platform the example is for. As we did with the posix threads.

Some device drivers which are for Linux is okay to use printf, because it's the default. But I would rather they report errors some way other.

Then there's the actual have_stdio usage, where we want to preformat some information like printing a conn table or ifinfo. That is more like a utility that's optional. And for this we do need a way of calling out to printf.

And finally the last usage i can think of is sscanf() for the rtable parser. But you know what? I think I want to remake that part somehow. And we have come a long way already with 90% of systems no longer requiring a routing table. Because an interface can be marked as the default. But when we do, i suggest we find a better way to store the data. Than pulling in sscanf.

yashi, 2021-12-19:
> The use of **FILE** is considered bad for build reproduction.

Yes.  But we are talking about _debug_ print.  Not logging print.  Wait, we only have `csp_print()`.... sigh...  Never mind about `__file__`.  My mind was in debug print because I was editing `csp_debug.[ch]`.

johandc, 2021-12-19:
Okay. But I see your point about the weak function requiring stdio.

So maybe if CSP print is enabled. But stdio is not, there should be an empty weak implementation.

PR #333 (2021-12-28): Make csp_print configurable
This PR makes `csp_print()` toggleable by `CSP_ENABLE_CSP_PRINT` option.  The main commit is "csp_print: Add option to toggle csp_print".  The others are minor fixes and changes.

One thing I'd also like to fix, but it's not done yet, is to have yet another option for `csp_print_func()` with `stdio.h`.  Because I think there is a situation where one want all `sscanf()` and `snprintf()` related functions in libcsp but don't really want to use `vprintf()` version of `csp_print_func()`.

I'm fine having an empty `csp_print_func()` for all arch except for the POSIX arch [... cut]
closed 2021-12-28 by johandc