libcsp/libcsp #322: python3 binding is broken

yashi, 2021-12-21:
With the commit 407e89661d7, `csp_socket_t` is no longer `csp_conn_t`.  This caused various parts of the bind to be broken.

johandc, 2021-12-21:
Oh yes

StephanAAU, 2022-05-06:
I'm guessing this is also related to this error below that I am seeing? Afaik, "address" has been moved away from he csp_conf_s struct?

```
../src/bindings/python/pycsp.c:105:47: error: ‘csp_conf_t’ {aka ‘struct csp_conf_s’} has no member named ‘address’
  105 |  if (!PyArg_ParseTuple(args, "bsss", &csp_conf.address, &csp_conf.hostname, &csp_conf.model, &csp_conf.revision)) {
```

johandc, 2022-05-11:
Yes, in CSP 2.0 each interface has it's own address. So there is no longer a global CSP node id.

gsantamarina, 2023-01-19:
Will this be fixed soon? Getting the same error when trying to compile the python3 bindings.

PR #411 (2023-07-18): Fix python bindings
This is based on #403 , but with better commit names and also the following:
* Changed argument parsing in pycsp to only work with primitive types and instead convert it to the fixed-width types defined in `stdint.h` after argument parsing (See https://github.com/libcsp/libcsp/commit/1f6a004fc2863dd48ee3a244c0173f60bc3f23e6 for details.)
* `pycsp_init` now accepts all the fields of `csp_conf_t` as arguments
* Adds a `csp_socket_close` for properly closing a socket.

Resolves #322
closed 2023-07-18 by yashi