libcsp/libcsp #335: Define __weak and it's friend

yashi, 2021-12-27:
We have 21 `__attribute__((weak))` in our code base.  It's easier if we have `__weak` as:

```c
#define __weak __attribute__((weak))
```

Or any varients
```c
#define __unused __attribute__((unused))
#define __maybe_unused __attribute__((unused))
#define __packed __attribute__((packed))
#define __noinit __attribute__((noinit))
```

johandc, 2021-12-27:
Is there ever any reason why __weak would be defined as anything else than the `gcc` style __attribute__((weak)) ??

I agree `__weak` is much prettier. Whoever came up with the `gcc` style of declaring attributes should be dragged somewhere and punished for his harm to mankind. It ain't pretty. But...

I actually prefer to avoid the define. Because it puts a requirement of the code, to always include a given header file. We see this already in some libraries, they include something called <compiler_util.h> or whatever. And this might be necessary when targeting multiple compilers. But for us, and if everybody understands the `gcc` style, i prefer to live without the additional header file.

yashi, 2021-12-27:
OK.  If you prefer without defines, I'm fine.  Closing.

closed 2021-12-27 by yashi

yashi, 2021-12-27:
> Is there ever any reason why __weak would be defined as anything else than the gcc style attribute((weak)) ??

No.