libcsp/libcsp #342: CMP route get/list

jgromes, 2022-01-24:
Hello everyone,

I'd like to ask whether there's some way to retrieve (or list) all routes currently configured on a node. There's the route set command, so it almost seems "natural" it could have a get counterpart.

Is this something that's planned for the future, or is it intentionally not included?

Thanks!

johandc, 2022-01-25:
Hello,

Good question. The route set came from a time with CSP 1 where there was only 32 addresses. And it only supports specific routes to a single node. There was no need for a list back tjen.

Next came the CIDR notation with netmask. Now it started to make sense to talk about a routing table. The way we chose to handle that was through the init API and let the user handle the routing setup at boot. We added a helper for parsing a string into a routing table. Cmp route set still remained as an emergency method of creating a route back out from any node back to the sender. This could be used if the routing table was misconfigured. But not as s primary configuration interface.

CSP does not handle configuration management. In general. There is no storage backend to write configuration to, other than RAM. So the best way to manipulate the routing table is to change the init parameters CSP is started with and reboot. Or write your own function/command to change it runtime.

In version 2.1 of CSP things are going to change drastically. Now a system is capable of having multiple network adresses. One per interface. And each interface had its own netmask too. This pretty much means the routing table isn't needed because it can calculate on which interface a node belongs.

CSP has the ifstat method. Buy you need to know the name of the interface to request it.

We ought to make a full interface list and routing table list as an output.

There is an active issue on how to manage routing tables on 2.1 and onwards.

jgromes, 2022-01-25:
Thanks for the detailed response!

> The route set came from a time with CSP 1 where there was only 32 addresses.

Probably should have specified, but yes, the question was concerned with CSP 1, as that's the version I'm familiar with. 

> Cmp route set still remained as an emergency method of creating a route back out from any node back to the sender

That's exactly what I like most about route_set, it can be used to manually re-route traffic in case of an emergency, which I think is one of the "killer features" of CSP in general.

However, I'm not sure about the statement that "CSP does not handle configuration management". I would argue that:
a) it requires some configuration to be stored
b) it provides read (if stats; clock) and write (cmp route set; clock) access to parts of said configuration - I don't think it matters that the changes done via cmp route set are volatile.
c) there is already a common command set in place (CMP, ping, reboot etc.)

I think all of it combined leads to some configuration management being performed by CSP. Following further, perhaps configuration management of CSP *should* be handled by CSP itself (or a protocol/command set above it, much like CMP). Otherwise, users have no choice but to implement their own configuration methods, and naturally, those will differ. That might not be a bad thing (some users will choose to handle configuration management on their own, even given the choice), however, I still like the idea of CSP being a completely self-contained package.

The above still applies to CSP 1 - I'm not familiar (yet) with CSP 2.x, so it might no longer be applicable.

Feel free to close this issue, it's just random thoughts on using CSP anyway ;)

johandc, 2022-01-25:
> "CSP does not handle configuration management"

What i mean by this is that, CSP relies on csp_init() and the csp_config struct being passed into the library at startup. Naturally you cannot configure every aspect of CSP over the air / network interface itself. Because if the list of interfaces wasn't setup there would be no method for talking to it.

After init, i think we have basically two configuration options:
  1) routing (we have cmp_route set only, at the moment)
  2) RDP options for new RDP connections (this should be moved into the API of the caller btw)

> B) It provides read access  + C) there is already a common command set in place.
Yes, but ping, buffree, reboot etc. are not examples of configuration management. That's telemetry output. Which is not what i am talking about.

> Clock
This is a chapter on its own. I would rather make a separation of the network library, and the time keeping functions. As a separate app that runs on top of CSP. This means it will also be decoupled from the release cycles of CSP, and allow for faster development of CSP apps. 

a) Any configuration that requires storage CSP expects to receive on csp_init()

I can recommend github.com/spaceinventor/libparam as a configuration management library :)

johandc, 2022-05-01:
Hello,

I believe this is solved by this related issue:
#334

I don't feel like we need a network faced interface for dynamic route configuration to be part of CSP right now. I mean, whats next, the address, network interface setup and deduplication? There are a lot of configuration and as i have mentioned before it should not be CSP's task to manage its own configuration.

One could argue that for parameters which needs to change after csp_init have been calleed. Like inserting a new route is an example of, it makes sense. And i tend to agree, this is why we will have csp route_set, because its part of a recovery method to always be able to configure a reverse route. So it's a kind of special function.

For normal operation, the routing table should be changed, and then CSP init called again. We typically configure our systems and reboot them once they are setup as we want them. And we don't touch routing tables while in operation.

We have multiple satellite radios, and the way we manage which one is active is not by CSP routing tables, so we always route packets to both radio interfaces. But we have a flag on each radio, to turn off or inhibit its transmitter. That we use for downlink selection.

Btw. here is an example on how we do some remote debugging of CSP configuration by using the csp_print feature

```c
#if (CSP_HAVE_STDIO)
static uint8_t _csp_print_cnf;
void csp_print_cnf_cb(struct param_s * param, int offset) {
	csp_iflist_print();
	csp_rtable_print();
	csp_iface_t * iface = csp_iflist_get_default();
	if (iface) {
		printf("  dfl if %s\n", iface->name);
	}
}
PARAM_DEFINE_STATIC_RAM(PARAMID_CSP_PRINT_CNF,    csp_print_cnf,     PARAM_TYPE_UINT8, 0, 0, PM_SYSCONF, csp_print_cnf_cb, "", &_csp_print_cnf, NULL);
#endif
```

closed 2022-05-01 by johandc