libcsp/libcsp #343: CFP2_SENDER_MASK

jsrmalvarez, 2022-02-03:
Hi, I'm playing with CSP over CAN.

I can't find any CFP 2.0 specification (except the code itself) so maybe this is not an issue.

In the definition of CFP2_SENDER_MASK (csp_if_can.h file, line 109)
shouldn't it be 0x3F instead of 0x1F, since CFP2_SENDER_SIZE is 6 bit?
Otherwise we are losing the MSb of the Sender ID field.

jsrmalvarez, 2022-02-07:
Thanks johandc for your reply.

Yes, the size of 6 bit is correct.

But the mask CFP2_SENDER_MASK is set to 0x1F.
The mask is taking only 5 bit. Shouldn't it be 0x3F to take the 6 bit?

johandc, 2022-02-08:
I was to quick to conclude and read MSB not MSb. Yes this is a bug.

jsrmalvarez, 2022-02-08:
My last phrase of the issue description was unnecesary and misleading.

johandc, 2022-05-01:
Fixed in develop

closed 2022-05-01 by johandc