libcsp/libcsp #346: CCSDS over CSP

PaulMadle, 2022-02-21:
Hi all, 

Apologise if this is not really the right forum to ask questions but I am trying to see if there is a way to integrate CCSDS protocol over the top of CSP. 
I am thinking that the CSP UDP service might offer a starting point. I am thinking that I could put the CCSDS data link layer TM/TC frames into a CSP UDP frame. The CCSDS would then take care of retransmissions of telecommands and file transfer in place of the RDP protocol defined in CSP. Does this sound feasible? 2 questions:
- What is the maximum data payload in bytes for a single CSP UDP frame?
- Does the CSP UDP frame include some kind of error detection (e.g. CRC).

I tried looking through the documentation and couldn't find answers to these questions.

Many thanks,

Paul.

johandc, 2022-02-22:
CSP 2.0 supports frame sizes of 2026 bytes on CAN bus, and higher on other links. Up to 16 bit length. So 65 KB.

CSP includes CRC32C.

johandc, 2022-02-22:
I don't see the reason why you would use CSP over a single link to encapsulate something else, unless the link you are using is CSP already.

CSP is used for routing on board and on ground. So if you don't use routing. Then CSP is just overhead.

PaulMadle, 2022-02-22:
Thanks @johandc... Much appreciated. The reason to consider this encapsulation is simply necessity due to the selection of otherwise incompatible systems. Agreed that it is an overhead. Thanks for the numbers :)

PaulMadle, 2022-02-23:
Many thanks for your wisdom @johandc, I am also trying to understand CSP 1.0 protocol rather than CSP 2.0. Is there a spec for this CSP 1.0 protocol?

johandc, 2022-02-23:
CSP 1 is a 4 byte header. It can be seen on the Wikipedia page of cubesat protocol. CSP 2 is 6 byte header and the spec can be found in the doc folder.

PaulMadle, 2022-02-23:
Fantastic, thankyou... I can see the header format for CSP 1.0 on https://en.wikipedia.org/wiki/Cubesat_Space_Protocol.
Is there something that specifies how this CSP message gets segmented into CAN frames for CSP 1.0?

I am aware that the first CAN frame in a sequence of CAN frames maps some of the fields in the CSP 1.0 header into the CAN 29 bit ID field and then some other fields into CAN data payload... Is there information on this for CSP 1.0?

johandc, 2022-02-23:
Yes that information is in csp_cfp.c / .h here on GitHub.

PaulMadle, 2022-02-23:
Brilliant, thankyou @johandc

closed 2022-03-04 by johandc