libcsp/libcsp #347: How to run older version of libcsp (v1.4) that still supports Windows?

mm16bn, 2022-03-08:
Hi, I believe this used to support Windows so I tried pulling an older version from the v1.4 tag. When I try to build the library with ./waf build install, it gives an extremely garbled output and fails. What is the last version where Windows is maintained? And what is the best configuration option for it?

johandc, 2022-03-09:
You need to go way back to 2012 I believe to the work Dan Erik made from the NUTS satellite project. I don't know why you would use windows, but if you have to, i would suggest updating the port on the latest version.

We have done away with a lot of the requirements on the OS. So it should actually be easier to port now. CSP only really needs semaphores and queues.

yashi, 2022-03-09:
@mm16bn I don't usually do Windows programming but the last time I cross built windows version was around c8446ee1fcfc5e99c25f544ef3fad24bfebc7bec.  You can see how I built it in the commit log.  Hope this helps.

mm16bn, 2022-03-25:
Thank you! This helps.

closed 2022-03-25 by mm16bn