libcsp/libcsp #350: Potential crash when adding more routes than defined by CSP_RTABLE_SIZE

valerijbielskij, 2022-04-01:
Following "if" statement found on csp_rtable_cidr.c:67 seems to be somehow broken/illogical.

```
if (rtable_inptr == CSP_RTABLE_SIZE)
  rtable_inptr = CSP_RTABLE_SIZE;
```

Since this is the only place where `rtable_inptr` is increased I'm assuming that the idea was to move this check one line above to prevent having this counter overflowing maximum table size.

johandc, 2022-04-06:
Thanks for that report
It should have been >= instead

valerijbielskij, 2022-04-07:
So the idea was to override the last entry in the list? Probably I would have expected that it would return CSP_ERR_INVAL in such case but I don't have strong opinion about it.

johandc, 2022-04-07:
Yes. I think the most recent entry should at least be accepted, in a safe manner. This way it should always be possible to setup a direct route in an emergency.

However I have recently gone over to using a simple array of numbers for the routing table. This makes it even simpler to configure and does not require any string work.

closed 2022-05-01 by johandc