libcsp/libcsp #360: CSP port handling

fjmolinas, 2022-04-15:
I'm trying to find more details on how CSP handles ports. I've not yet found a document with clear description, but from the [wikipedia entry](https://en.wikipedia.org/wiki/Cubesat_Space_Protocol) and the code itself I can see that:

- ports 0-7 are "reserved", but this applies if the receiver actually calls the `default_handler`, the Wikipedia entry also does not say if this applied to 2.0.

- ports 8-47 should be free to use, but the socket and callback bind options will only allow binding to "any" or to one of the ports from 0-7,  but I can't bind to a specific port from 8-47, only filter the ports of interests later.

- the wiki entry says that ports 48-63 are ephemeral ports for outgoing connections, but the code actually sets all connections ports to `CSP_PORT_MAX_BIND` + i, which is actually ports in the range from 17 to CSP_CON_MAX.

I understand the need for the reserved ports, but I don't see why other ports would need to be specifically used for outgoing/incoming connections. It also means that If I wan to use port 47 I actually need to have `CSP_PORT_MAX_BIND` be of size 47, which seems like a waste of memory.

In a branch, I re-worked a bit this and added the `port` value to the `csp_port_t` structure, this way bind ports can still be kept limited but allowing to cover to full scope of allowed ports. Is this a valid change to propose or am I breaking something?

Also is there a reason for keeping a distinction between outgoing and incoming connection ports? I would prevent me from for example expecting responses on the same port I sent a message.

If I've missed documentation explaining all of this I would be happy to be referred to it.

Thanks!

johandc, 2022-04-15:
Okay, this is a good question that nobody has actually asked before.

So let's add that to the documentation starting with this response.

First some definitions for CSP 2:

0 to 7 reserved for CSP services (i wanted to move all CSP services to a single port, but it is what it is for legacy)
8 to MAX_BIND for incoming
MAX_BIND + 1 to MAX_PORTS is outgoing
MAX_PORTS + 1 is the CSP_ANY port, which can catch all incoming traffic that does not hit an assigned port.

Okay so what is the reason for the split between incoming and outgoing. There is just one reason and that is efficiency. An array lookup is a constant time O(1) operation. Walking a linked list of ports, or any other binding method is O(n) or whatever search operation you throw at the problem. But an array of port numbers is also a waste of good memory if not all ports are used. Therefore in general try to stick to lower port numbers for incoming. Don't make huge gap, because that will also waste outgoing ports due to the split.

Now the outgoing ports they can loop the connection table and check if any given port is in use. This is a O(n*m) operation until it finds a free outgoing port number. When it finds a free port in the connection table it also needs to look for that in the bound ports tavle. So not very fast.
However if all outgoing ports are sequentially assigned (i.e no incoming ports in between) we can always assume the next free port is going to be the last plus one. We still do the check in case there is one long running connection in use ln that number, but it reduces it to O(n) for 99% of the cases.

fjmolinas, 2022-04-19:
@johandc thanks for the answer! Some follow up questions

> 0 to 7 reserved for CSP services (i wanted to move all CSP services to a single port, but it is what it is for legacy)
> 8 to MAX_BIND for incoming
> MAX_BIND + 1 to MAX_PORTS is outgoing
> MAX_PORTS + 1 is the CSP_ANY port, which can catch all incoming traffic that does not hit an assigned port.

AFAIK `MAX_PORTS` is not defined in the tree though.

> Okay so what is the reason for the split between incoming and outgoing. There is just one reason and that is efficiency. 

This then means that there are no "real" restrictions on how outgoing/bind ports are assigned if it is dependent on `CSP_PORT_MAX_BIND`. Other than the ports reserved for CSP services from an implementation point, I could assign them as I would like, there is nothing in the routing protocol that depends on this.

johandc, 2022-04-19:
1. Yed. CSP_PORT_ANY is actually defined to max bind port plus one. Indeed there is no max port define.

2. No you cannot assign source ports. CSP connect does that. And in fact the latest version have fixed source ports for each connection in the conn table, now that they are statically allocated.

closed 2022-05-01 by johandc