libcsp/libcsp #366: Individual server and client is not working in libcsp 1

sankariselvaraj, 2022-04-28:
hi i m trying to receive packets to server from client,i used libcsp1,but i am not able to receive.I used libcsp1 and took inputs from wiki from this git page.
i can give these inputs
/* Define subsystem address to 0 */                          in clientserver program
#define MY_ADDRESS  0
/* Init CSP with address MY_ADDRESS */               in  clientserver program
csp_init(MY_ADDRESS);
/* Setup default route to CAN interface */
csp_route_set(CSP_DEFAULT_ROUTE, &csp_can_tx, CSP_HOST_MAC);                    in src/csp_init.c

/* Start router task with 500 word stack, OS task priority 1 */
csp_route_start_task(500, 1);                                        in  clientserver program
and i am confused where to give other inputs?
please reply back me that where could i add
these are the inputs
1./* CAN configuration struct for SocketCAN interface "can0" */
struct csp_can_config can_conf = {.ifc = "can0"};
2./* Init buffer system with 10 packets of maximum 320 bytes each */
csp_buffer_init(10, 320); - if i use this in src/buffer.c gtting segmentation error
3./* Init the CAN interface with hardware filtering */
csp_can_init(CSP_CAN_MASKED, &can_conf)
4./* Init CSP with address MY_ADDRESS */
csp_init(MY_ADDRESS);             where it should be client server program or any other inner files?

server code
#include <stdio.h>
#include <string.h>
#include <unistd.h>
#include <stdlib.h>

#include <csp/csp.h>
#include <csp/arch/csp_thread.h>
#include <csp/drivers/usart.h>
#include <csp/drivers/can_socketcan.h>
#include <csp/interfaces/csp_if_zmqhub.h>
#include <csp/interfaces/csp_if_can.h>
/* Server port, the port the server listens on for incoming connections from the client. */
#define MY_SERVER_PORT		10
#define HAVELIBSOCKETCAN       1

/* Commandline options */
static uint8_t server_address = 255;

/* test mode, used for verifying that host & client can exchange packets over the loopback interface */
static bool test_mode = false;
static unsigned int server_received = 0

 CSP_DEFINE_TASK(task_server) {
    /* Create socket without any socket options */
    csp_socket_t *sock = csp_socket(CSP_SO_NONE);

    /* Bind all ports to socket */
    csp_bind(sock, CSP_ANY);

    /* Create 10 connections backlog queue */
    csp_listen(sock, 10);

    /* Pointer to current connection and packet */
    csp_conn_t *conn;
    csp_packet_t *packet;

    /* Process incoming connections */
    while (1) {
        /* Wait for connection, 10000 ms timeout */
        if ((conn = csp_accept(sock, 10000)) == NULL)
            continue;

        /* Read packets. Timout is 1000 ms */
        while ((packet = csp_read(conn, 1000)) != NULL) {
            switch (csp_conn_dport(conn)) {
                case MY_PORT:
[... 429 lines cut ...]
    

    /* Start client thread */
    if ((server_address != 255) ||  /* server address specified, I must be client */
        (default_iface == NULL)) {  /* no interfaces specified -> run server & client via loopback */
        csp_thread_create(task_client, "CLIENT", 1000, NULL, 0, NULL);
    }

    /* Wait for execution to end (ctrl+c) */
    while(1) {
        csp_sleep_ms(3000);

        if (test_mode) {
            /* Test mode is intended for checking that host & client can exchange packets over loopback */
            if (server_received < 5) {
                csp_log_error("Server received %u packets", server_received);
                exit(1);
            }
            csp_log_info("Server received %u packets", server_received);
            exit(0);
        }
    }

    return 0;
}

johandc, 2022-04-28:
Go to GitHub spaceinventor and csh. That is the current best example of CSP usage that I have got.

closed 2022-04-28 by johandc

sankariselvaraj, 2022-04-28:
Hi,thanku for your support.But actually I need to work in libcsp 1 alone for my application,so other possibilities so please can u tell me the files where to put these inputs ?
1./ CAN configuration struct for SocketCAN interface "can0" /
struct csp_can_config can_conf = {.ifc = "can0"};
2./ Init buffer system with 10 packets of maximum 320 bytes each /
csp_buffer_init(10, 320); - if i use this in src/buffer.c gtting segmentation error
3./ Init the CAN interface with hardware filtering /
csp_can_init(CSP_CAN_MASKED, &can_conf)
4./ Init CSP with address MY_ADDRESS */
csp_init(MY_ADDRESS); where it should be client server program or any other inner files?

johandc, 2022-05-01:
If you want to run CSP 1. You should go back to csp v1.6, or use latest development and set csp.version = 1 before calling csp_init().

1. There is an example on setting up the socketcan interface in the csp_yaml.c file
2. There is an example of setting up the buffers in csh.c
3. Maked configuration is not supported on linux/socketcan yet. (i think)
4. We don't use global address since CSP 2. It is assigned to network interfaces now.

sankariselvaraj, 2022-05-01:
Hi
Thanku for your reply.Actually we were working for real time project under
satellite system .I could nt see v1.6 because the page not found.And also i
could nt not find csp_yaml.c and csh.c.Anf where should this csp_init()
called?
A.inside the task server
B.inside the csp_ buffer.h
C inside the csp.h
And also can u please provide the link for the above v1.6 and csp_yaml and
cash c files.
Regards
Sankari selvaraj