libcsp/libcsp #369: ACK unconditionally accepted via RST + ACK

martinerikwerner-aac, 2022-05-07:
If a RST + ACK segment arrives in the OPEN state, the ACK is accepted unconditionally:

https://github.com/libcsp/libcsp/blob/ba15bf68a78c1397aaaa7673426ff9ccb44bad35/src/csp_rdp.c#L525-L531

regardless if the ACK is inside the current expected window, or if the received segment has a sequence number that is outside the expected window (which would cause the RST part to be ignored)

Is this intentional?

It seems like this would mean that if a receiver with:

```
state == OPEN;
snd_una == 1234;
rcv_cur == 2345;
```

would receive a segment with:

```
flags == RST | ACK;
seq_nr == 0;
ack_nr == 0;
```

it would then accept this as a valid ACK and set `snd_una = 1` and then ignore the RST as being out of sequence.

It does not seem like this is a thing that it should be accepting in this case?

johandc, 2022-05-11:
That sounds correct.
I'm not sure the code would ever send a reset + ack though.
This needs to be investigated further.

martinerikwerner-aac, 2022-05-11:
ACK + RST are being used in these places at the moment:

https://github.com/libcsp/libcsp/blob/ba15bf68a78c1397aaaa7673426ff9ccb44bad35/src/csp_rdp.c#L791

https://github.com/libcsp/libcsp/blob/ba15bf68a78c1397aaaa7673426ff9ccb44bad35/src/csp_rdp.c#L966

https://github.com/libcsp/libcsp/blob/ba15bf68a78c1397aaaa7673426ff9ccb44bad35/src/csp_rdp.c#L552

johandc, 2024-09-18:
After closer inspection, This is not an issue.

Whenever a packet is received the ACK number should always be stored in SND_UNA. This is because we use any packet, wether it is a data packet, a dedicated ack packet or in fact a RST packet to always let the other party know what the RCV_CUR. It never hurts to update this value on any incoming packet.

There is no reason not to believe the sender of the packet.

johandc, 2024-09-18:
Closing this. Feel free to reopen if i am wrong.