libcsp/libcsp #373: Libcsp compilation for MCU

satzguyz, 2022-06-26:
Hello Experts,

I am new for the libcsp source code. I can download and compile it for Ubuntu using waf build (default option). It is getting compiled using native gcc.

I want to compile this code for ST MCU series using arm-gcc compiler. Any thoughts on where and how to do helps ?

yashi, 2022-06-26:
You can use, Waf, Meson, or CMake, which ever you know how to use with your compiler.

- Waf: https://waf.io/book/#_cross_compilation_concerns
- Meson: https://mesonbuild.com/Cross-compilation.html
- CMake: https://cmake.org/cmake/help/book/mastering-cmake/chapter/Cross%20Compiling%20With%20CMake.html

satzguyz, 2022-06-26:
@yashi Thanks for the clarification. If I understood right, do I need to write my compilation scripts for cross-compiling it?

johandc, 2022-06-26:
I recommend meson.

With meson you first take whatever ST has for hello world example and build that using meson. If theres an old Makefile hurry up and get rid of that. Then write the build with meson or cmake to make your life much easier. - Next you include csp as a subproject into your meson using the meson dependency system. An example can be found on github.com/spaceinventor/csh - allbeit that's a linux build. But setting up cross compilation for meson is as simple as adding a cross_file.txt with the path to the compiler.

closed 2022-06-28 by johandc