libcsp/libcsp #384: CSP on FreeRTOS w/o POSIX

urblockhead01, 2022-08-17:
Nice to meet you 

I'm trying to get ISiS-OBC interact with QubeAIS, one of AIS Receiver, using CSP.
ISIS-OBC, aka IOBC, is running on FreeRTOS without POSIX.

Because of absent of POSIX, I don't think the port name such as "/dev/ttyUSB0" can be assigned for CSP configuration.
Without port name and POSIX, it seems not to be able to get File Descriptor using `open(fd, ~~~.~~)` function.
So what I've tried is to replace `read` in `usart_rx_thread` or `write` in `csp_usart_write`  with the equivalent UART read/write function involved in ISIS OBC Library

Of course I failed to get it work 😢.
Do you think there's any way to make CSP work on freeRTOS system without POSIX? 

thank you.

(Link for QubeAIS : https://www.satlab.com/products/qubeais/)

yashi, 2022-08-18:
You don't need POSIX subsystem to use libcsp on FreeRTOS.  I haven't had time to add GitHub Actions but we have FreeRTOS sample repo at https://github.com/libcsp/libcsp-freertos.  Hope it helps.

If you have any question, please post it here with exact steps you've done.  That'll help us help you.  Thanks.

urblockhead01, 2022-08-24:
I have two more question

1. libcsp really doesn't require POSIX on FreeRTOS?
In [meson.build] at [https://github.com/libcsp/libcsp-freertos](https://github.com/libcsp/libcsp-freertos), source file seems to consist of POSIX

2. How did you set csp device without termios?
As far as I know, a posix contains termios and termios must be used to address the port, such as "/dev/ttyUSB0". Then, how can I specify a csp device without posix?

urblockhead01, 2022-08-24:
> You don't need POSIX subsystem to use libcsp on FreeRTOS. I haven't had time to add GitHub Actions but we have a FreeRTOS sample repo at https://github.com/libcsp/libcsp-freertos. Hope it helps.
> 
> If you have any question, please post it here with exact steps you've done. That'll help us help you. Thanks.

I have two more question

libcsp really doesn't require POSIX on FreeRTOS?
In [meson.build] at https://github.com/libcsp/libcsp-freertos, source file seems to consist of POSIX

How did you set csp device without termios?
As far as I know, a posix contains termios and termios must be used to address the port, such as "/dev/ttyUSB0". Then, how can I specify a csp device without posix?

yashi, 2022-08-27:
What physical media do you want your OBC to talk CSP on?

CSP _can_ be used on UART and Ethernet.  To use them, a Linux port need some accessor functions, `read()` / `write()` and `send()` / `recv()`, respectively.  And that's POSIX.

Zephyr RTOS can run libcsp on a CanBus, and it uses its own accessor functions and its [canbus device driver](https://docs.zephyrproject.org/latest/hardware/peripherals/canbus/controller.html).  A port of your FreeRTOS needs some accessor and a device driver for your physical media.

> 1. libcsp really doesn't require POSIX on FreeRTOS?

No. It doesn't require POSIX.
OTOH, the libcsp-freertos _is_ a [port of FreeRTOS on Linux](https://www.freertos.org/FreeRTOS-simulator-for-Linux.html).  And it uses Linux and POSIX.  But libcsp proper does not require POSIX nor Linux.

> 2. How did you set csp device without termios?

termios is a part of POSIX/Unix [TTY interface](https://man7.org/linux/man-pages/man4/tty.4.html). If you don't use TTY, you don't use termios.  I don't use it when I run libcsp on Ethernet.

urblockhead01, 2022-08-30:
Let me make another question plz and I think it would be better if you ignore everything I asked. 
Sorry for getting you confused.

> What physical media do you want your OBC to talk CSP on?

If I understand the question correctly, I want CSP communication between ISIS OBC and QubeAIS via UART.

ISIS OBC(IOBC) has its own UART driver. And that's why I replaced `read()`, `write()` function with `UART_read()`, `UART_write()`
Thanks to IOBC's UART driver, I don't need TTY interface as you said. 

For better description of the problem, I wrote down what I did in sequence below.
----
# Test on Ubuntu 
1. First, downloaded open source for CSP communication with [QubeAIS](https://www.satlab.com/products/qubeais/), called [satctl](https://satlab.com/software/satctl/)
2. For testing satctl, I connected Ubuntu laptop and QubeAIS. At this time, USB for Ubuntu and RS-232 for QubeAIS was used.(Of course signal between two device was converted using USB to RS232 converter)
3. I erased all the useless code in satctl project except lib/csp, client/ais and src/satctl.c
4. I modified src/satctl.c and set `csp address = 1, node = 22, type = TYPE_UART, device = "/dev/ttyUSB0", bitrate = 115200`
5. The wafscript was modified to configure modified satctl project.
7. I ran satctl on the Ubuntu laptop and send ping with `csp_ping(22, 1000, 100, CSP_O_NONE)` and it works 
____
# Implement on IOBC
8. with the same code as I used in step 1~7, I edited `csp_uart_open()` in lib/csp/src/drivers/usart/usart_linux.c to initialize UART port with IOBC's own UART driver ( The code seems not to be opened because of security issue. sorry )
9. And then, I replaced `write()` in `csp_usart_write()` with `UART_write()` which is the part of IOBC UART driver
10. Also, replaced `read()` in `usart_rx_thread()` with `UART_read()`
11. Finally, some input arguments such as stack size, priority (for functions related to creating task(thread)) get assigned with `1024`, `configMAX_PRIORITIES-1` each
12. After connecting IOBC with QubeAIS via RS-232, I ran satctl on IOBC and send ping with `csp_ping(22, 1000, 100, CSP_O_NONE)`. 
____
HOWEVER, It seems not to be work.
13. To check if `UART_write()` send packet, I added below code to print packet data. If `UART_write()` doesn't work, only first data would be printed 
[image]

The result imply that `UART_write()` works! 
[image]

14. I checked if `UART_read()`,  in `usart_rx_thread()`, works with this code. And UART seems not to read anything. 
[image]

15. At this time, I send reboot command using `csp_reboot()` from IOBC to QubeAIS to know whether `UART_write()` really works
16. But, the reboot command didn't make QubeAIS to do anything

In this case, what do you think my problem is?

johandc, 2022-09-01:
Seems to be a problem with the UART driver or the physical layer then. :)
If you go and verify what is transmitted on the wires with an oscilloscope, i bet the issue will be here.

Best Regards,

urblockhead01, 2022-09-01:
> Seems to be a problem with the UART driver or the physical layer then. :) If you go and verify what is transmitted on the wires with an oscilloscope, i bet the issue will be here.
> 
> Best Regards,

Holy shit.....
Oscilloscope got me know that rx/tx wiring has been wrong...

The most important is the basic...😂🤣😅
Thank you .. !!