libcsp/libcsp #385: csp_buffer_get does not set the length field of the packet buffer it returns.

Taylorr82, 2022-09-26:
It's strange to have to set the length of the packet manually when the function to return a packet has a size parameter

johandc, 2022-09-30:
The size parameter is not meant to be inserted, it is meant as a maximum size of the allocation. The size field should rather be removed, now that CSP buffers are fixed size.

Taylorr82, 2022-09-30:
Interesting, they've been a fixed size, statically allocated since at least version 1.6 (the one we're using on our project). I can make a commit here that will remove the size field if you'd like. 

Perhaps there should be an abstraction function to set the length field of the packet, does that exist already in this version?

johandc, 2022-11-12:
Fixed

closed 2022-11-12 by johandc