libcsp/libcsp #387: USART exit behaviour (v1.6)

jeffstjean, 2022-10-14:
The USART driver is documented that it will call `exit()` on read failures. Is there a reason for this behavior? It makes it tricky to recover from since the parent process will exit. Is there any side effects if we change this from an `exit()` to a `csp_thread_exit()` aside from limiting which platforms we can compile for (referencing the "Not supported on all platforms." note)? 

Why doesn't the `usart_rx_thread()` function just continue in the loop if errors occur? We have virtual serial ports setup using socat and killing the socat process causes the whole application running CSP to die.

yashi, 2022-10-16:
I'm sure your're talking about `usart_linux.c`.  I believe there is not big reason to call `exit()`.  Most of the errors from `read(2)` are programming errors or hardware errors and can be avoided if you've done right or unable to workaround, respectively.  But I believe we should handle `EAGAIN`, though.

> Is there any side effects if we change this from an exit() to a csp_thread_exit()

It depends on the error you are getting.  If the device has a hardware error, you can't recover from restarting the RX thread.  You can't even re-start CSP driver layer because you won't be able to open the device. (this is your case, your program can't start `socat` by itself but have to wait ptys to be up).

> Why doesn't the usart_rx_thread() function just continue in the loop if errors occur?

Because it might give you infinite errors and takes 100 % CPU time.
What is the point of calling `read()` again after it returned an error?

Oh, wait.  There are some drivers calling `read()` again after an error, ie socket can.

> We have virtual serial ports setup using socat and killing the socat process causes the whole application running CSP to die.

This is more like having a hardware error and kernel removed the device from it. In that case, we can't just restart rx thread but the Linux device driver must be restarted (assuming the hardware error is transient).
Why do you want your CSP to keep running after your PTYs are gone?

jeffstjean, 2022-10-16:
The main concern here is that the driver error causes our entire application to terminate due to the `exit()` call. This is of course a scenario we wish to avoid but the application handles other mission critical tasks. To give more context, we are running Core Flight System (cFS) and one of the cFS applications (which are run as threads) is responsible for setting up CSP along with the USART driver. Ideally, the CSP RX thread and the cFS application thread terminate but the application does not so it can restart the cFS application and reinitialize CSP. This doesn't appear to be possible with the given implementation since it will terminate the process and not just thread.

yashi, 2022-10-18:
I can see your pain point.

You can call `csp_usart_open()` without `rx_call_back`.  In this case `csp_usart_open()` does not create a rx_thread.  You can call your own rx function.

If you are using `csp_usart_open_and_add_kiss_interface()`, replace it with yours.

WDYT?

johandc, 2022-10-20:
I think you should create your own UART task that tries to recover gracefully.

We have the same problem with socketcan actually. If we remove the can dongle from our PC the linux device disappears, but the driver keeps reading from a non existing file descriptor. The entire application needs to be re initialised because it does not have a reopen/reinit device features.

jeffstjean, 2022-10-25:
> I think you should create your own UART task that tries to recover gracefully.

If I'm not mistaken, this would require running it as a separate process or doing something weird with signal dispatching since it calls `exit()` directly.

> You can call `csp_usart_open()` without `rx_call_back`.

That's most likely going to be our solution here as it's the least intrusive.

closed 2022-10-25 by jeffstjean

johandc, 2022-10-29:
Of course you should not call exit() - you should make a pull request with the improvements that enable the uart interface to fail gracefully. :)