libcsp/libcsp #388: SocketCAN does not work outside of promiscuous mode

ghvisser, 2022-10-27:
### Description
When using csp_can_socketcan_open_and_add_interface to create a CAN interface there is no option to specify the interface address. The filter then defaults to address 0 (through calloc). Receiving messages is then stuck at the "read" function in socketcan_rx_thread, as this thread is started immediately. The thread then gets stuck as no message is received sent to address 0. Modifying the iface.addr after the call to csp_can_socketcan_open_and_add_interface therefore has no effect.

### To Reproduce
Open a csp socket through csp_can_socketcan_open_and_add_interface, without promiscuous mode. Try to modify the address through modifying the iface. Observe no packets are received and that execution of the socketcan_rx_thread is stuck.

### Expected behavior
In addition to the promiscuous mode flag, the function should have an address as its arguments.

### Additional context
N/A

johandc, 2022-10-29:
Yes, we have this support for the embeded drivers. See the example driver for atmel ASF in the examples folder. So far no-one has cared to implement filtered version for linux. Since linux usually is mostly used on PC's and laptops in the lab. And has enough ressources to do software based filtering.

I will mark this as a feature request. Which you are welcome to work on if you wish :)

Zosoworld, 2023-05-26:
In my case, I am also working in development with SocketCAN and Linux on an amd64 computer, for software which will later be cross-compiled to ARM Linux. 
I've tried the client-server example using the `vcan0` interface:

```bash
ip link add dev vcan0 type vcan
ip link set vcan0 up
```
and no packet is received at all:

```bash
Initialising CSPINIT CAN: device: [vcan0], bitrate: 0, promisc: 0
Connection table
[00 0x562dbe683960] S:0, 0 -> 0, 0 -> 0 (17) fl 0
[01 0x562dbe683a70] S:0, 0 -> 0, 0 -> 0 (18) fl 0
[02 0x562dbe683b80] S:0, 0 -> 0, 0 -> 0 (19) fl 0
[03 0x562dbe683c90] S:0, 0 -> 0, 0 -> 0 (20) fl 0
[04 0x562dbe683da0] S:0, 0 -> 0, 0 -> 0 (21) fl 0
[05 0x562dbe683eb0] S:0, 0 -> 0, 0 -> 0 (22) fl 0
[06 0x562dbe683fc0] S:0, 0 -> 0, 0 -> 0 (23) fl 0
[07 0x562dbe6840d0] S:0, 0 -> 0, 0 -> 0 (24) fl 0
Interfaces
LOOP       addr: 0 netmask: 14 mtu: 0
           tx: 00000 rx: 00000 txe: 00000 rxe: 00000
           drop: 00000 autherr: 00000 frame: 00000
           txb: 0 (0B) rxb: 0 (0B) 

CAN        addr: 0 netmask: 0 mtu: 248
           tx: 00000 rx: 00000 txe: 00000 rxe: 00000
           drop: 00000 autherr: 00000 frame: 00000
           txb: 0 (0B) rxb: 0 (0B) 

Route table
0/0 CAN
Server task started
```
whereas using loopback it works as expected.
Even though I've tried in my own code with promiscuous mode in the `server`, I have not been able to establish communication through `vcan0`, even though I can see the packets being sent with `candump`:
```bash
$ candump -a vcan0
  vcan0  101FE022   [8]  00 00 14 C0 00 01 02 03   '........'
  vcan0  101FE025   [1]  04                        '.'
  vcan0  101FE042   [8]  00 00 15 00 00 01 02 03   '........'
  vcan0  101FE045   [1]  04                        '.'
```

johandc, 2023-06-01:
Try setting the promisc flag to 1
That should work for vcan

marckleinebudde, 2023-06-15:
The `promisc` flag doesn't have any effect on CAN. If you see CAN frames with `candump` on `vcan0` your application should see them, too.

Zosoworld, 2023-06-15:
I have tried with `promisc` to true with no avail. While testing, I have seen that when compiling with pure CMake, the `CSP_HAVE_LIBSOCKETCAN` variable is not properly set and so I had to do it in the code. I am running the `csp_server_client` example, with just a `#define CSP_HAVE_LIBSOCKETCAN 1` and `promisc = true`, and compiled with the following `CMakeLists.txt`:

```cmake
cmake_minimum_required(VERSION 3.9)
project(libcsp-vcan-test CXX C)

include(FetchContent)

set(CMAKE_EXPORT_COMPILE_COMMANDS ON)

FetchContent_Declare(libcsp
  GIT_REPOSITORY https://github.com/libcsp/libcsp.git
  GIT_TAG 9114bbd
)

FetchContent_GetProperties(libcsp)
if(NOT libcsp_POPULATED)
  FetchContent_Populate(libcsp)
  add_subdirectory(${libcsp_SOURCE_DIR} ${libcsp_BINARY_DIR} EXCLUDE_FROM_ALL)
endif()

set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++17 -Wno-psabi -Wall -Wextra -O3 -fno-exceptions")

add_executable(libcsp-vcan-test csp_server_client.c csp_server_client_posix.c)
target_link_libraries(libcsp-vcan-test PRIVATE libcsp)
```

The result is still the same, that I can see the frames with `candump`, but not with the software:
```bash
./libcsp-vcan-test -c vcan0 -a 10
Initialising CSP
INIT CAN CSP
Connection table
Interfaces
Route table
Server task started
^C
```

```bash
./libcsp-vcan-test -c vcan0 -r 10
Initialising CSP
INIT CAN CSP
Connection table
Interfaces
Route table
Client task startedPing address: 10, result -1 [mS]
reboot system request sent to address: 10
Ping address: 10, result -1 [mS]
reboot system request sent to address: 10
^C
```

marckleinebudde, 2023-06-28:
start the application that should be receiving things with strace: `strace -ff -o /tmp/log <APPLICATION> <APPLICATION_ARGS>`

Zosoworld, 2023-07-04:
I apologize for the delay, here are the logs for both the client and server, just in case.
[logs.tar.gz](https://github.com/libcsp/libcsp/files/11943470/logs.tar.gz)

marckleinebudde, 2023-07-04:
Looking at the code: 

https://github.com/libcsp/libcsp/blob/12326d109e32672e83a95e728b9579a66ee47802/src/drivers/can/can_socketcan.c#L103-L129

we see that libcsp has the concept of `promisc` mode, too. I updated https://github.com/libcsp/libcsp/issues/388#issuecomment-1592724580 to clarify my confusion.