libcsp/libcsp #389: Received CAN messages with size close to the MTU get dropped on receive

ghvisser, 2022-11-09:
In csp_if_can.c CSP V2 CAN message reception starts writing at packet->frame_begin, which is offset by -6 bytes from packet->data. The overflow check uses packet->frame_length, dlc and iface->mtu to determine overflow. This fails to compensate for the -6 bytes offset. This means that the actual MTU is 7 bytes less than the iface->mtu. Packets get sent properly at the MTU length, but dropped on reception. Only packets that are 6 bytes under the MTU get sent properly.

Proposed fix is subtracting 6 from the calculated packet size, as CSP_ID2_HEADER_SIZE from csp_id.c is not visible in csp_if_can.c. There is also plenty of other code that has implicit assumptions on the header size. I will attach a pull request. If this needs to be solved more systematically this define needs to be moved to the header file.

johandc, 2022-11-12:
This should be fixed with the latest merge

closed 2022-11-12 by johandc

ghvisser, 2022-11-15:
Good fix, thank you.