libcsp/libcsp #395: What is currently the most up-to-date stable branch?

gsantamarina, 2023-01-24:


yashi, 2023-01-25:
It depends.  We technically do not have "stable" branches.  But basically both v1 and v2.0 branch receives "fixes" only nowadays.  If you find any problem, please let us know.

Here is a list of the branches we have.

- `develop` : The development branch.  All new things should go here.  And this is the default branch.  This will be 2.1 or something else.
- `libcsp-1` : libcsp version 1.x
- `libcsp-2-0` : libcsp version 2.0
- `master` : This one might be confusing.  But it points to `libcsp-1` and we leave it there for backward compatibility.

#244 might gives you some idea as well.

gsantamarina, 2023-01-25:
Thanks Yashi for the quick answer.
Is it safe to use the `develop` branch? 
What features/important bug-fixes are present in `develop` that are missing from `libcsp-2-0`?

yashi, 2023-01-26:
> Is it safe to use the develop branch?

It depends.  It's certainly safe to run it on your board. I shouldn't blow up. But I can't ensure that it's perfectly safe to run it for your mission critical human related project.

What features/important bug-fixes are present in develop that are missing from libcsp-2-0?

```
git rev-list --no-merges origin/libcsp-2-0..origin/develop | wc -l
87
```

There are only 87 non-merge commits between. :wink:

To read them, run the following
```
gitk origin/libcsp-2-0..origin/develop
```

closed 2023-01-31 by yashi