libcsp/libcsp #396: Running libcsp on STM32 without FreeRTOS

Evidlo, 2023-01-25:
I was wondering if it's possible to run libcsp on a system without an RTOS.

i.e. We would like to just call some libcsp tick function periodically in a timer interrupt to update the state machine and provide hooks for handling data RX and TX.

Is this realistic at all, or is divorcing libcsp from an RTOS non-trivial?

yashi, 2023-01-26:
libcsp does not support a bare metal configuration right now.

To run libcsp on a board, you need to 1) initialize the SoC, 2) setup C runtime environment, 3) load your executable image, 4) prepare threads and stacks, 5) setup interrupt and timer, 6) initialize peripherals you need, 6) etc.   That's basically a new OS.

I don't say you can't do it.  It's definitely possible.  But I don't see the point of implementing a new OS, instead of using an existing OS.

Evidlo, 2023-01-26:
Thanks for the response.

closed 2023-01-26 by Evidlo

MoSamArafat, 2024-10-14:
Hi, hope all is well with everyone who sees this!

I would like to ask if there is any development work since the original post towards running CSP bare-metal?

Thank you :)

yashi, 2024-10-15:
no.