libcsp/libcsp #397: csp_can_pbuf_cleanup does not work for ISR

sdhunna21, 2023-02-11:
https://github.com/libcsp/libcsp/blob/b2996d29b492e5d1dcaafc4cc01ab2367d540a11/src/interfaces/csp_if_can_pbuf.c#L48 calls 

https://github.com/libcsp/libcsp/blob/develop/src/interfaces/csp_if_can_pbuf.c#L70

When csp_can_pbuf_cleanup is called from an ISR, it should check the task_woken flag, e.g.:
`uint32_t now = (task_woken) ? csp_get_ms_isr() : csp_get_ms();`

Instead it always calls csp_get_ms, this causes an error in port.c, line 543 which is in vPortEnterCritical (when libcsp is compiled with FreeRTOS).
```
void vPortEnterCritical( void )

/* This is not the interrupt safe version of the enter critical function so
	assert() if it is being called from an interrupt context.  Only API
	functions that end in "FromISR" can be used in an interrupt.  Only assert if
	the critical nesting count is 1 to protect against recursive calls if the
	assert function also uses a critical section. */
	if( ulCriticalNesting == 1 )
	{
		configASSERT( ulPortInterruptNesting == 0 );
	}
```

johandc, 2023-02-11:
Thanks for reportin this. It was already fixed in the latest space inventor master branch. I will perform a merge right away.
https://github.com/spaceinventor/libcsp/commit/f5fec5ed1fefef69f6750955f2076de499d61cf5

closed 2023-02-11 by johandc

sdhunna21, 2023-02-13:
I'm not seeing the change in develop.  Which branch was it merged to?  I'm using CSP with header version 1.x on can bus, should I be using the develop branch?

johandc, 2023-02-16:
Oh, I found it's already part of a pull request that is blocked for other reasons. So you will have to use the spaceinventor/master temporarily for the fix.