libcsp/libcsp #399: build example file

spaceorwhat, 2023-05-08:
I'm not able to build the example file properly to make the csp_server_client executable. I've tried building the example directory multiple times but i the executable is not generated. Please help T_T

yashi, 2023-05-09:
Care to share how you built? Please show us the entire commands you entered as well as the output.

spaceorwhat, 2023-05-09:
Okay so here's what I've done so far
-I changed python to python3 in the buildall.py file 
-used cmake and ninja to build as given in the install.md file
cmake -GNinja -B builddir
cd builddir
ninja
CMake Warning (dev) at CMakeLists.txt:1 (project):
  cmake_minimum_required() should be called prior to this top-level project()
  call.  Please see the cmake-commands(7) manual for usage documentation of
  both commands.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- No build type selected, default to MinSizeRel
-- Looking for include file sys/socket.h
-- Looking for include file sys/socket.h - found
-- Looking for include file arpa/inet.h
-- Looking for include file arpa/inet.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.2") 
-- Checking for one of the modules 'libzmq'
-- Checking for one of the modules 'libsocketcan'
-- Configuring done (0.5s)
-- Generating done (0.0s)
-- Build files have been written to: /home/srishti/penguin/csp/libcsp-develop/builddir
[40/40] Linking C shared library liblibcsp.so

in the examples directory i gave-
cmake ..
got this output:
CMake Warning (dev) at CMakeLists.txt:1 (project):
  cmake_minimum_required() should be called prior to this top-level project()
  call.  Please see the cmake-commands(7) manual for usage documentation of
  both commands.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- No build type selected, default to MinSizeRel
-- Looking for include file sys/socket.h
-- Looking for include file sys/socket.h - found
-- Looking for include file arpa/inet.h
-- Looking for include file arpa/inet.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.2") 
[... 95 lines cut ...]

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.

This warning is for project developers.  Use -Wno-dev to suppress it.

-- Generating done (0.0s)
-- Build files have been written to: /home/srishti/penguin/csp/libcsp-develop/examples/build
ninja: Entering directory `build'
ninja: error: unknown target 'examples/csp_server_client'
Traceback (most recent call last):
  File "/home/srishti/penguin/csp/libcsp-develop/examples/./buildall.py", line 70, in <module>
    main(args.build_system, rest)
  File "/home/srishti/penguin/csp/libcsp-develop/examples/./buildall.py", line 58, in main
    build_with_cmake()
  File "/home/srishti/penguin/csp/libcsp-develop/examples/./buildall.py", line 32, in build_with_cmake
    subprocess.check_call(cmake_compile + targets)
  File "/usr/lib/python3.9/subprocess.py", line 373, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['ninja', '-C', 'build', 'examples/csp_server_client', 'examples/csp_arch', 'examples/zmqproxy']' returned non-zero exit status 1.

Now i have no clue what to do further :(

yashi, 2023-05-09:
### Using buildall.py
If you are happy using `buildall.py`, all you need to do is to run it.

```
$ examples/buildall.py
'distclean' finished successfully (0.000s)
Setting top to                           : /home/yashi/src/libcsp 
Setting out to                           : /home/yashi/src/libcsp/build 
Checking for 'gcc' (C compiler)          : /usr/lib/ccache/gcc 
Checking for program 'python'            : /usr/bin/python 
Checking for header stdio.h              : yes 
Checking for header sys/socket.h         : yes 
Checking for header arpa/inet.h          : yes 
'configure' finished successfully (0.078s)
Waf: Entering directory `/home/yashi/src/libcsp/build'
[ 1/37] Compiling src/interfaces/csp_if_can.c
[ 2/37] Compiling src/csp_hex_dump.c
[ 3/37] Compiling src/crypto/csp_hmac.c
[ 4/37] Compiling src/arch/posix/csp_semaphore.c
[ 5/37] Compiling src/arch/posix/csp_clock.c
[ 6/37] Compiling src/csp_init.c
[ 7/37] Compiling src/crypto/csp_sha1.c
[ 8/37] Compiling src/interfaces/csp_if_kiss.c
[ 9/37] Compiling src/csp_id.c
[10/37] Compiling src/csp_sfp.c
[11/37] Compiling src/interfaces/csp_if_can_pbuf.c
[12/37] Compiling src/csp_dedup.c
[13/37] Compiling src/csp_service_handler.c
[14/37] Compiling src/arch/posix/pthread_queue.c
[15/37] Compiling src/csp_rtable_cidr.c
[16/37] Compiling src/csp_crc32.c
[17/37] Compiling src/csp_rdp_queue.c
[18/37] Compiling src/arch/posix/csp_queue.c
[19/37] Compiling src/csp_qfifo.c
[20/37] Compiling src/csp_buffer.c
[21/37] Compiling src/csp_port.c
[22/37] Compiling src/arch/posix/csp_system.c
[23/37] Compiling src/interfaces/csp_if_udp.c
[24/37] Compiling src/csp_iflist.c
[25/37] Compiling src/csp_services.c
[26/37] Compiling src/interfaces/csp_if_i2c.c
[27/37] Compiling src/csp_debug.c
[28/37] Compiling src/csp_route.c
[29/37] Compiling src/arch/posix/csp_time.c
[30/37] Compiling src/csp_promisc.c
[31/37] Compiling src/csp_bridge.c
[32/37] Compiling src/csp_rtable_stdio.c
[33/37] Compiling src/interfaces/csp_if_lo.c
[34/37] Compiling src/csp_conn.c
[35/37] Compiling src/csp_rdp.c
[36/37] Compiling src/csp_io.c
[37/37] Linking build/libcsp.a
Waf: Leaving directory `/home/yashi/src/libcsp/build'
'build' finished successfully (0.086s)
'distclean' finished successfully (0.002s)
Setting top to                           : /home/yashi/src/libcsp 
Setting out to                           : /home/yashi/src/libcsp/build 
Checking for 'gcc' (C compiler)          : /usr/lib/ccache/gcc 
Checking for program 'python'            : /usr/bin/python 
Checking for header stdio.h              : yes 
Checking for header sys/socket.h         : yes 
Checking for header arpa/inet.h          : yes 
Checking for program 'pkg-config'        : /usr/bin/pkg-config 
Checking for 'libsocketcan'              : yes 
Checking for 'libzmq'                    : yes 
'configure' finished successfully (0.083s)
Waf: Entering directory `/home/yashi/src/libcsp/build'
[ 1/48] Compiling src/drivers/can/can_socketcan.c
[ 2/48] Compiling src/csp_hex_dump.c
[ 3/48] Compiling src/crypto/csp_hmac.c
[... 139 lines cut ...]
src/crypto/edit_cache: phony
src/crypto/rebuild_cache: phony
src/drivers/edit_cache: phony
src/drivers/rebuild_cache: phony
src/interfaces/edit_cache: phony
src/interfaces/rebuild_cache: phony
examples/edit_cache: phony
examples/rebuild_cache: phony
csp_arch: phony
csp_server_client: phony
driver_can: phony
if_zmq: phony
libcsp: phony
zmqproxy: phony
all: phony
build.ninja: RERUN_CMAKE
clean: CLEAN
help: HELP

$ ninja csp_server_client
[3/3] Linking C executable examples/csp_server_client

$ ls examples
CMakeFiles  cmake_install.cmake  csp_server_client
```

yashi, 2023-05-09:
BTW, there is a package called `python-is-python3`.

```
sudo apt install python-is-python3
```

Might saves you a second.

spaceorwhat, 2023-05-09:
I installed the zmq library and python-is-python3 and the below command worked for me:
cmake -GNinja -S . -B builddir
Thank you so much
But now the problem I'm facing is after I give the ./csp_server_client in the directory where its present, without any options, the cursor just blinks and nothing happens. Is it supposed to take a long time or something?

yashi, 2023-05-10:
Ah, thanks.  That's because `CSP_PRINT_STDIO` is not set with CMake. Moreover, you have to manually change `0` to `1` in `CMakeLists.txt` with the current `develop` branch.

The variable should be in cache so that we can [set it with `-D`](https://cmake.org/cmake/help/latest/command/set.html#set-cache-entry).

```
cmake -S . -B builddir -DCSP_PRINT_STDIO=1 
```

The following change should be applied.

```diff
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 2c89c44..ef0cb54 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -18,7 +18,7 @@ endif()
 
 set(CSP_HAVE_STDIO 1)
 set(CSP_ENABLE_CSP_PRINT 1)
-set(CSP_PRINT_STDIO 0)
+set(CSP_PRINT_STDIO 0 CACHE BOOL "Use C standard I/O for csp_print")
 set(CSP_QFIFO_LEN 15 CACHE STRING "Length of incoming queue for router task.")
 set(CSP_PORT_MAX_BIND 16 CACHE STRING "Length of incoming queue for router task")
 set(CSP_CONN_RXQUEUE_LEN 16 CACHE STRING "Number of packets in connection queue")
```

And also for PC the value should be `1` for the default.

spaceorwhat, 2023-05-10:
Okay thank you so much that worked. Thanks for all the help.

closed 2023-05-10 by spaceorwhat