libcsp/libcsp #409: When building a C++ program, need to add extern "C" block around libcsp includes.

negatratoron, 2023-07-10:
Compiler: g++ 11.3.0

Expected behavior: C++ programs compile and link against libcsp just by including the appropriate header files and linking the object files.

Actual behavior: Get the linker error ``undefined reference to `csp_init()'``

---

Reproducible example: In a new project directory:

Clone libcsp into `lib/csp` (as done here: https://github.com/spaceinventor/csh)

meson.build file:
```
project('TestProgram',
	'cpp',
	subproject_dir: 'lib'
)

csp_dep = dependency('csp', fallback: ['csp', 'csp_dep'], required: true)

main = executable(
	'main',
	[ 'src/main.cpp' ],
	dependencies: [csp_dep]
)
```

src/main.cpp file.  Program compiles and links after `extern "C"` lines uncommented:
```
// extern "C" {
#include <csp/csp.h>
// }

int main() {
  csp_init();

  return 0;
}
```

yashi, 2023-07-12:
@negatratoron Please try the PR against real C++ programs.  And let me know how it works.

negatratoron, 2023-08-03:
Thanks, the issue does not reproduce with commit [7e6d365](https://github.com/yashi/libcsp/commit/7e6d3657144c101cc4958257edf4c35866288562) @yashi

closed 2023-08-03 by yashi