libcsp/libcsp #415: waf breaks when python is python3

Buzzzz, 2023-07-27:
On a system where python is python3 (e.g. debian bookworm), the waf 2.0.18 as distributed with libcsp fails with the following message:

```
zip2: Data integrity error when decompressing.
        Input file = t.bz2, output file = t

It is possible that the compressed file(s) have become corrupted.
You can use the -tvv option to test integrity of such files.

You can use the `bzip2recover' program to attempt to recover
data from undamaged sections of corrupted files.

bunzip2: Deleting output file t, if it exists.
Error: Waf cannot be unpacked, check that bzip2 support is present
```

Bzip2 support is present (/bin/bzip2 and friends).

Updating the waf script to the current version (2.0.25) fixes this problem.

yashi, 2023-07-27:
I can't reproduce it.

```
╭─ yashi@x570 ──────────────────────────────────────────────────────────── /tmp ─╮
╰─ 20:18:23 ❯ cd libcsp                                                         ─╯
╭─ yashi@x570 develop ───────────────────────────────────────────── /tmp/libcsp ─╮
╰─ 20:18:26 ❯ ./waf configure build                                             ─╯
Setting top to                           : /tmp/libcsp 
Setting out to                           : /tmp/libcsp/build 
Checking for 'gcc' (C compiler)          : /usr/lib/ccache/gcc 
Checking for program 'python'            : /usr/bin/python 
Checking for header stdio.h              : yes 
Checking for header sys/socket.h         : yes 
Checking for header arpa/inet.h          : yes 
'configure' finished successfully (0.086s)
Waf: Entering directory `/tmp/libcsp/build'
[ 1/37] Compiling src/interfaces/csp_if_can.c
[ 2/37] Compiling src/csp_hex_dump.c
[ 3/37] Compiling src/crypto/csp_hmac.c
[ 4/37] Compiling src/arch/posix/csp_semaphore.c
[ 5/37] Compiling src/arch/posix/csp_clock.c
[ 6/37] Compiling src/csp_init.c
[ 7/37] Compiling src/crypto/csp_sha1.c
[ 8/37] Compiling src/interfaces/csp_if_kiss.c
[ 9/37] Compiling src/csp_id.c
[10/37] Compiling src/csp_sfp.c
[11/37] Compiling src/interfaces/csp_if_can_pbuf.c
[12/37] Compiling src/csp_dedup.c
[13/37] Compiling src/csp_service_handler.c
[14/37] Compiling src/arch/posix/pthread_queue.c
[15/37] Compiling src/csp_rtable_cidr.c
[16/37] Compiling src/csp_crc32.c
[17/37] Compiling src/csp_rdp_queue.c
[18/37] Compiling src/arch/posix/csp_queue.c
[19/37] Compiling src/csp_qfifo.c
[20/37] Compiling src/csp_buffer.c
[21/37] Compiling src/csp_port.c
[22/37] Compiling src/arch/posix/csp_system.c
[23/37] Compiling src/interfaces/csp_if_udp.c
[24/37] Compiling src/csp_iflist.c
[25/37] Compiling src/csp_services.c
[26/37] Compiling src/interfaces/csp_if_i2c.c
[27/37] Compiling src/csp_debug.c
[28/37] Compiling src/csp_route.c
[29/37] Compiling src/arch/posix/csp_time.c
[30/37] Compiling src/csp_promisc.c
[31/37] Compiling src/csp_bridge.c
[32/37] Compiling src/csp_rtable_stdio.c
[33/37] Compiling src/interfaces/csp_if_lo.c
[34/37] Compiling src/csp_conn.c
[35/37] Compiling src/csp_rdp.c
[36/37] Compiling src/csp_io.c
[37/37] Linking build/libcsp.a
Waf: Leaving directory `/tmp/libcsp/build'
'build' finished successfully (0.092s)
╭─ yashi@x570 develop ───────────────────────────────────────────── /tmp/libcsp ─╮
╰─ 20:18:30 ❯ python --version                                                  ─╯
Python 3.11.4
╭─ yashi@x570 develop ───────────────────────────────────────────── /tmp/libcsp ─╮
╰─ 20:18:32 ❯ python3 --version                                                 ─╯
Python 3.11.4
╭─ yashi@x570 develop ───────────────────────────────────────────── /tmp/libcsp ─╮
╰─ 20:18:34 ❯ lsb_release -a                                                    ─╯
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux trixie/sid
Release:	n/a
Codename:	trixie
╭─ yashi@x570 develop ───────────────────────────────────────────── /tmp/libcsp ─╮
╰─ 20:19:00 ❯                                                                   ─╯
```

Buzzzz, 2023-07-27:
Interesting! I dug a bit into the issue.

I've checked out a clean copy of libcsp and everything works.
The dirty copy I had came from a CentOS 7 onto a Debian Bookworm.
This must have been a local cache effect. Whatever data waf had cached on CentOS has confused the waf on Debian (e.g. paths to tools like bunzip2). Running the new waf-2.0.25 cache data was recreated and everything worked as bunzip2 was found.
So lets close this as a side effect of moving files between incompatible Linux Distros.

Buzzzz, 2023-07-27:
Solved after throwing away cached waf state.

closed 2023-07-27 by Buzzzz