libcsp/libcsp #423: MacOSX support dropped but not documented

dmopalmer, 2023-08-01:
MacOSX and Windows support have been dropped as of 2.0, according to the `CHANGELOG`.
```
- removed: Windows Support (moved to contrib, maintainer needed)
- removed: Mac OSX Support (moved to contrib, maintainer needed)
```
However the `README.md` still indicates that they are supported.

Is there a build procedure that can compile the targets in contrib (with the caveat that the code is unmaintained, and maybe not supporting all three of `waf`, `meson` and `cmake`)?

Is it possible to checkout the 1.6 version (`git checkout  libcsp-1`) and compile it for the Mac?  I did this but the python library wasn't generated.

Should the `README.md` be changed to indicate (footnoted?) that these targets will not be available until a maintainer is found?

From my particular project, an ad-hoc specialized python implementation of CSP is the selfish-least-resistance path, but that incurs technical debt, doesn't test interoperability, and doesn't support the ecosystem.

yashi, 2023-08-02:
Hi,

The code for OSX is moved to contrib at a5256340b6d8c1fcfd344e8ebcafea32d7971a8e. Everything left in the `develop` and `libcsp-2-0` branches is something we forgot to remove.

Theoretically, what you need to do is to move code under contrib to the original location and update the API to match the current one. Unfortunately, we don't have any instructions to do that.

We are happy to take PRs to bring Mac support back to the current branch, but I'll merge a cleanup PR which removes Mac OS bits from the develop if no one is willing to step up.

dmopalmer, 2023-08-04:
I am working on this.  The major consequences are:

-  `<csp/csp_endian.h>` is back but it is just an include of the standard `<endian.h>`, except: 
    - On Mac: include `<machine/endian.h>` and `#define` the `htobe16()` etc. functions to alias macos equivalents.
- `# include <stdlib.h>` to define `malloc()` instead of the non-standard `<malloc.h>`
    - This should probably be done with or without the Mac, unless it affects another target   
- `__attribute__((section(".noinit")))` will need to be `#defined` with a slight change (or drop?) on the Mac
- `pthread_queue_t` needs to be made to work on the Mac.

These should have no as-compiled effect on the other targets, but a slight uglification of the code. Also, I do not promise to be a permanent maintainer.  

Is this enough to keep Mac on the future roadmap of libcsp, or do you want to strip it out as unmaintained?

yashi, 2023-08-04:
Since the current maintainers don't use MacOS at all, at least a build test on GitHub Actions is need to merge the MacOS port.  Otherwise we can't keep it buildable.  Would you mind port port `.github/workflows/build-test.yml` as well?

- `endian.h`: it seems we need something like https://github.com/mikepb/endian.h/blob/master/endian.h
- `malloc.h` is non-standard header.  I'll address it in #429.
- `.noinit` attributes: Are you using Clang?  I have [a PR ](https://github.com/libcsp/libcsp/pull/422)with clang build test and it seems working on Linux.  Or should we just use `__attribute__((noinit))` instead of `__attribute__((section(".noinit")))`?
- `csp_queue`: You don't need to use pthread as you might already know.  All you have to do is to port `csp_queue` API to MAC OSX. (`DispatchQueue`?)

yashi, 2023-08-04:
- `endian.h`: simple one is fine I suppose.  I'll take a look at it once you create a PR
- `.noinit` is ELF specific. I guess we have to create something like `__noinit`macro in Linux kernel or Zephyr RTOS, which uses `attributes` on supported platforms and becomes empty on non-supported platforms like MacOS
- `csp_queue`: No I don't have any for MacOS.  The one we used to had was based on pthread. But, if you know any platform specific queue implementation, it shouldn't be that hard to create a new one.  Zephyr version is less than 100 lines.  FreeRTOS's is less than 60.

yashi, 2023-08-04:
> I haven't used the Github Actions before. Do I just uncomment `#- macos-latest` on line 11 of `.github/workflows/build-test.yml`

I assume so.  You can always test it on your fork.  (There is a tool called "[act](https://github.com/nektos/act)" but it might take a bit of time to set it up)

dmopalmer, 2023-08-08:
I got it compiled on the Mac (so syntax is at least correct).

At this point, should I submit the PR, or wait until I have actually tested it?

I enabled the python bindings:
```
meson setup --wipe builddir -D enable_python3_bindings=true
cd builddir
ninja
```
But when I run python to start testing, it says that the canbus stuff isn't there.
```
----> 1 import libcsp_py3 as libcsp

ImportError: dlopen(/Users/palmer/repo/notmine/libcsp/builddir/libcsp_py3.cpython-311-darwin.so, 0x0002): symbol not found in flat namespace '_csp_can_socketcan_open_and_add_interface'
```

I don't see the `can_socketcat.c.o` file in the final link.  Should `pycsp.c` have `pycsp_can_socketcan_init` conditionally-compile based on `CSP_HAVE_LIBSOCKETCAN` (as it does, e.g. for `#if CSP_HAVE_LIBZMQ`).

I have done this and copied the `python_bindings_example_*.py` into the buld dirctory for testing.  The server dies because I don't have `zmq`. and the client:
```
    print("CMP ident:", libcsp.cmp_ident(options.server_address))
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
csp.Error: csp_cmp_ident(), result/error: -3
```

Is the python stuff supposed to work, as of pulling the upstream `develop` branch this afternoon?

yashi, 2023-08-08:
Please tell me how to reproduce the error you are seeing.

I've done

```
$ rm -rf builddir && meson setup builddir -Denable_python3_bindings=true
The Meson build system
Version: 1.2.0
Source dir: /home/yashi/src/libcsp
Build dir: /home/yashi/src/libcsp/builddir
Build type: native build
Project name: csp
Project version: 2.0
C compiler for the host machine: sccache cc (gcc 13.2.0 "cc (Debian 13.2.0-1) 13.2.0")
C linker for the host machine: cc ld.bfd 2.41
Host machine cpu family: x86_64
Host machine cpu: x86_64
Found pkg-config: /usr/bin/pkg-config (1.8.1)
Run-time dependency yaml-0.1 found: YES 0.2.5
Library c found: YES
Run-time dependency threads found: YES
Run-time dependency libsocketcan found: YES 0.0.12
Run-time dependency libzmq found: YES 4.3.4
Has header "sys/socket.h" : YES 
Has header "arpa/inet.h" : YES 
Configuring csp_autoconfig.h using configuration
Program python3 found: YES (/usr/bin/python3)
Run-time dependency python3 found: YES 3.11
Run-time dependency python found: YES 3.11
meson.build:101: WARNING: Project targets '>=0.53.2' but uses feature introduced in '0.63.0': python_installation.extension_module with implicit dependency on python. use python_installation.dependency()
Build targets in project: 5
WARNING: Project specifies a minimum meson_version '>=0.53.2' but uses features which were added in newer versions:
 * 0.63.0: {'python_installation.extension_module with implicit dependency on python'}

csp 2.0

  User defined options
    enable_python3_bindings: true

Found ninja-1.11.1 at /usr/bin/ninja

$ ninja -C builddir
$ env PYTHONPATH=build python
```
Then

```
Python 3.11.4 (main, Jun  7 2023, 10:13:09) [GCC 12.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import libcsp_py3
>>> 
```

yashi, 2023-08-25:
@dmopalmer How is it coming? If you are still working on it, I'm fine waiting for it. But if you are not, would you mind if I merge #424 for the time being? I'll, of course, revert it when you've ported it to MacOS. Or, you can include documentation update in your PR.

dmopalmer, 2023-08-25:
I have it compiling (with one of the build systems) but I am having trouble testing it with Python. It doesn’t seem to make an installable package that can be `pip —install`ed, so `import` only works when `cd`ed to a specific directory.  Should there be a python package built?Other priorities mean that I won’t be able to work on it for at least another week, so proceed accordingly.Sent from my iPad

johandc, 2023-09-09:
Hey guys, just wanted to give you a thumbs up for working on this :+1:

dmopalmer, 2024-04-03:
(Returning after a long absence from this).
I just tried building under linux to avoid any Mac-specific problems, and the python module is not installed in the standard path.
``` bash
conda activate test
cmake -DCSP_ENABLE_PYTHON3_BINDINGS=ON -GNinja -B builddir
cd builddir
ninja
```
When I run `python` in the `conda activate test` environment in the `builddir`, it loads the `.so` library from the `builddir` (but fails because I don't have `libzmq` or `libsocketcan`).
```python
Python 3.12.2 | packaged by Anaconda, Inc. | (main, Feb 27 2024, 17:35:02) [GCC 11.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import libcsp_py3
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: /storage/work/dmp6407/repo/notmine/libcsp/builddir/libcsp_py3.cpython-312-x86_64-linux-gnu.so: undefined symbol: csp_can_socketcan_open_and_add_interface
```

But the library is not installed anyplace else in the path, so running python from any other directory will fail.
```
>>> import libcsp_py3
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'libcsp_py3'
```

Is this the expected/accepted behavior?  Is manually moving the `.so` library to the a directory in the path the next step of the build?

yashi, 2024-04-04:
I don't use `conda` and don't know what it does but this is my case:

```
❯ git describe
v1.6-1052-gfac3bda1
❯ cmake -DCSP_ENABLE_PYTHON3_BINDINGS=ON -GNinja -B builddir
- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/lib/ccache/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/lib/ccache/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- No build type selected, default to MinSizeRel
-- Looking for include file sys/socket.h
-- Looking for include file sys/socket.h - found
-- Looking for include file arpa/inet.h
-- Looking for include file arpa/inet.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.8.1") 
-- Checking for one of the modules 'libzmq'
-- Found /usr/lib/x86_64-linux-gnu/libzmq.so 4.3.5
-- Checking for one of the modules 'libsocketcan'
-- Found /usr/lib/x86_64-linux-gnu/libsocketcan.so 0.0.12
-- Found Python3: /usr/include/python3.11 (found version "3.11.8") found components: Development Development.Module Development.Embed 
-- Configuring done (0.5s)
-- Generating done (0.0s)
-- Build files have been written to: /home/yashi/src/libcsp/builddir
❯ cd builddir
❯ ninja
[44/44] Linking C shared module libcsp_py3.cpython-311-x86_64-linux-gnu.so
❯ find | grep py3
./libcsp_py3.cpython-311-x86_64-linux-gnu.so
./CMakeFiles/libcsp_py3.dir
./CMakeFiles/libcsp_py3.dir/src
./CMakeFiles/libcsp_py3.dir/src/bindings
./CMakeFiles/libcsp_py3.dir/src/bindings/python
./CMakeFiles/libcsp_py3.dir/src/bindings/python/pycsp.c.o
❯ python
Python 3.11.8 (main, Mar 26 2024, 12:26:07) [GCC 13.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import libcsp_py3
>>> dir(libcsp_py3)
['CSP_ANY', 'CSP_BUF_FREE', 'CSP_CMP', 'CSP_ERR_AGAIN', 'CSP_ERR_ALREADY', 'CSP_ERR_BUSY', 'CSP_ERR_CRC32', 'CSP_ERR_DRIVER', 'CSP_ERR_HMAC', 'CSP_ERR_INVAL', 'CSP_ERR_NOBUFS', 'CSP_ERR_NOMEM', 'CSP_ERR_NONE', 'CSP_ERR_NOTSUP', 'CSP_ERR_RESET', 'CSP_ERR_SFP', 'CSP_ERR_TIMEDOUT', 'CSP_ERR_TX', 'CSP_ERR_USED', 'CSP_FCRC32', 'CSP_FFRAG', 'CSP_FHMAC', 'CSP_FRDP', 'CSP_MAX_TIMEOUT', 'CSP_MEMFREE', 'CSP_NO_VIA_ADDRESS', 'CSP_O_CRC32', 'CSP_O_HMAC', 'CSP_O_NOCRC32', 'CSP_O_NOHMAC', 'CSP_O_NONE', 'CSP_O_NORDP', 'CSP_O_RDP', 'CSP_PING', 'CSP_PRIO_CRITICAL', 'CSP_PRIO_HIGH', 'CSP_PRIO_LOW', 'CSP_PRIO_NORM', 'CSP_PS', 'CSP_REBOOT', 'CSP_SO_CONN_LESS', 'CSP_SO_CRC32PROHIB', 'CSP_SO_CRC32REQ', 'CSP_SO_HMACPROHIB', 'CSP_SO_HMACREQ', 'CSP_SO_NONE', 'CSP_SO_RDPPROHIB', 'CSP_SO_RDPREQ', 'CSP_UPTIME', 'Error', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__spec__', 'accept', 'bind', 'buffer_free', 'buffer_get', 'buffer_remaining', 'can_socketcan_init', 'close', 'cmp_clock_get', 'cmp_clock_set', 'cmp_ident', 'cmp_peek', 'cmp_poke', 'cmp_route_set', 'conn_dport', 'conn_dst', 'conn_sport', 'conn_src', 'connect', 'get_hostname', 'get_model', 'get_revision', 'init', 'kiss_init', 'listen', 'packet_get_data', 'packet_get_length', 'packet_set_data', 'ping', 'print_connections', 'print_routes', 'rdp_get_opt', 'rdp_set_opt', 'read', 'reboot', 'recvfrom', 'route_start_task', 'rtable_check', 'rtable_clear', 'rtable_load', 'rtable_set', 'send', 'sendto', 'sendto_reply', 'service_handler', 'shutdown', 'socket', 'transaction', 'zmqhub_init']
>>> 
```

BTW, this is not MacOSX related. So it'd be better to open a new issue.

PainOchoco, 2024-05-30:
Hello any news for the MacOS port?
On my end, for the python bindings to be imported I need to explicitly add `/usr/local/lib` to the path like so:

```py
import sys
sys.path.append("/usr/local/lib")
```

Then I can `import libcsp_py3 as libcsp...`

PR #424 (2023-08-25): Remove Mac OS X leftover
This fixes #423