libcsp/libcsp #425: API for Per interface CSP Address

yashi, 2023-08-02:
We've removed CSP address from csp_conf at 0eda8e1d85717 and have moved to per interface CSP addresses.  But APIs are not yet ready.

This issue is an ambrela issue to collect interface API changes for CSP address.  Known fixes for variaous interfaces ATM

- CAN: 0d4dac28a5bf75390bdd1ef26e1c109094e0a032
- ZMQ: #404

yashi, 2023-09-28:
`csp_iface_t` (or `struct csp_iface_s`) is an open struct, meaning we can access members of it any time.  Do we really need a function parameter for `node_id`?

We can already do without `node_id`.

```c
csp_usart_open_and_add_kiss_interface(..., &iface);
iface->addr = 1;
```

If so, why don't we pass `netmask` and others?

johandc, 2023-09-28:
I dont even like the fact that CSP calls malloc. It would have rather had the helper function took some "empty" struct pointers, that i could have allocated statically in advance before calling the helper function. But with these, helpers (which is only used in CSH / linux context anyways) we use malloc. And it is a little messy to say the least, what is set afterwards (addr and netmask is added manually after all the helpers in CSH) and which ones have been added as arguments to be set inside the helper function.

johandc, 2023-09-28:
I think the best would be to allow the user to setup the struct with the interface config, and the driver config. And call the init function with the struct. Without any allocation happening on the inside.

johandc, 2023-09-28:
So that would mean to remove the helper arguments almost completely and eventually simplify those helpers away.

yashi, 2024-11-11:
For now all interface leant interface address parameter. I'm sure it'd be nice to have option struct but that can be implmented in the future. I'll close this for now.