libcsp/libcsp #433: Tag v2.0

yashi, 2023-08-05:
It's been long over due to release version 2.0.

tniessen, 2023-08-05:
Hi @yashi! We are using libcsp 1. When libcsp 2.0 is tagged, will it be considered to be at least as stable as libcsp 1, or should we wait before upgrading?

yashi, 2023-08-05:
Software in general do have many bus in .0 releases that's because users tend to wait until it's released. Libcsp is no exception I'd say.

Please help us squash as many bus as possible by using it!

Thanks

michalkaptur, 2024-04-17:
Hi @yashi, do you plan on releasing v2 anytime soon? What would be the right version to use until then, [develop](https://github.com/libcsp/libcsp/tree/develop) or [libcsp-2-0](https://github.com/libcsp/libcsp/tree/libcsp-2-0)?

I couldn't find the branching policy neither in the readme nor in the docs.

yashi, 2024-04-18:
Taken from https://github.com/libcsp/libcsp/issues/395#issuecomment-1403199937

> - `develop` : The development branch. All new things should go here. And this is the default branch. This will be 2.1 or something else.
> - `libcsp-1` : libcsp version 1.x
> - `libcsp-2-0` : libcsp version 2.0
> - `master` : This one might be confusing. But it points to libcsp-1 and we leave it there for backward compatibility.

And, Yes, we should document this in the README.  Thanks.

yashi, 2024-04-19:
`v2.0` is just tagged.

closed 2024-04-19 by yashi