libcsp/libcsp #438: "./waf install" does not install any files or directories

Buzzzz, 2023-08-09:
The docs suggests that "waf install" should install files to the directory specified at configuration time with --prefix. This appears to not work. No files are installed and the prefix directory is not created.

Can anyone confirm that this is not just an effect on my system (debian bookworm)? To reproduce:

```
$ ./waf configure --prefix=/tmp/csp
Setting top to                           : /home/username/gitprj/libcsp
Setting out to                           : /home/username/gitprj/libcsp/build
Checking for 'gcc' (C compiler)          : /bin/gcc
Checking for program 'python'            : /bin/python
Checking for header stdio.h              : yes
Checking for header sys/socket.h         : yes
Checking for header arpa/inet.h          : yes
'configure' finished successfully (0.235s)
$ ./waf build
Waf: Entering directory `/home/username/gitprj/libcsp/build'
[ 1/37] Compiling src/interfaces/csp_if_can.c
[ 2/37] Compiling src/csp_hex_dump.c
...
[36/37] Compiling src/csp_port.c
[37/37] Linking build/libcsp.a
Waf: Leaving directory `/home/username/gitprj/libcsp/build'
'build' finished successfully (0.299s)
$ ./waf install
Waf: Entering directory `/home/username/gitprj/libcsp/build'
Waf: Leaving directory `/home/username/gitprj/libcsp/build'
'install' finished successfully (0.024s)
$ ls -l /tmp/csp
ls: cannot access '/tmp/csp': No such file or directory
```

yashi, 2023-08-10:
Try `waf configure --prefix=/tmp/csp --install-csp`.

I'm not sure why we have this but it works.

Buzzzz, 2023-08-10:
Yashi-san, I can confirm that adding `--install-csp` works. Needing the extra argument is a bit surprising. If it stays that way, `doc/INSTALL.md` should be updated because the instructions currently provided (`./waf build install`) do not work. BTW, I also tried the meson build, it does install as expected.

yashi, 2023-08-10:
I agree that the option is suprising.  If @johandc doesn't object, I can make a PR to remove the option.  @johandc, what do you think?

johandc, 2023-09-09:
I think its because waf doesnt really carry the differentiation between when libcsp is compiles as a static library and when its compiled as a standalone, for the python bindings and libcsp.so. I dont really use waf any longer, so i cannot argue for or against having this option.

I belive that when libcsp is built as a static library we do not want to install the libcsp.a to the /usr/local/lib directory.

yashi, 2023-11-19:
I've done some research, and it appears that Waf distinguishes between `build` and `install` through `ctx.is_install`.

`InstallContext` is a simple, thin wrapper around `BuildContext` with `ctx.is_install` set to `waflib.Build.INSTALL`. In both `waf build` and `waf install`, the `build` function is called once. If you run `waf build install`, the `build` function will be called twice.

Therefore, in my conclusion, we can safely replace `--install-csp`, or more precisely, `if ctx.options.install_csp`, with `if ctx.is_install`.

PR #492 (2023-11-22): waf: Replace --install-csp with waf install
The option --install-csp was introduced in the commit ebe19dc49.  But waf can distiguish build and install via ctx.is_install.

https://waf.io/apidocs/Build.html#waflib.Build.BuildContext.is_install

Thus, this commit replaces --install-csp with waf install.

This closes #438.
closed 2023-11-22 by yashi