libcsp/libcsp #451: buildall.py fails in container

danieldavidson, 2023-09-19:


yashi, 2023-09-19:
#413

danieldavidson, 2023-09-19:
Is `apt-get install`-ing `libzmq3-dev` not enough to overcome the linked issue?

danieldavidson, 2023-09-19:
I realized that the Dockerfile that is successful, works because I'm only linking the `csp_server_client` target. `csp_server_client` is only the first of three targets in `buildall.py`, but it's enough to run the example. Linking `zmqproxy` fails with the same errors about undefined references that I pasted above.

yashi, 2023-09-20:
> Is `apt-get install`-ing `libzmq3-dev` not enough to overcome the issue mentioned in #413? All three Dockerfiles include `libzmq3-dev`.

Oops.  You are right.

Do you have `pkg-config` installed?

The line:

```
: && /usr/bin/cc -Os -DNDEBUG  examples/CMakeFiles/zmqproxy.dir/zmqproxy.c.o -o examples/zmqproxy  -Wl,-rpath,/libcsp/build  libcsp.so && :
```

should be like:

```
: && /usr/bin/cc -Os -DNDEBUG  examples/CMakeFiles/zmqproxy.dir/zmqproxy.c.o -o examples/zmqproxy  -Wl,-rpath,/home/yashi/src/libcsp/build  libcsp.so  -lzmq && :
```

Because you don't have `-lzmq` part, it fails.
</details>

danieldavidson, 2023-09-20:
That solved it. I appreciate your help! 🙏

PR #452 (2023-09-20): CMake: Print Library Search Status
Unlike `find_package()`, `pkg_search_module()` doesn't indicate whether it has found the requested libraries or not. Since we do not strictly require these libraries, we don't set them as `REQUIRED`.

This commit modifies CMake to either print the discovered binary path and its version or indicate not found.

It should help in a case like #451.

This closes #451.
closed 2023-09-20 by danieldavidson