libcsp/libcsp #463: Can't use ZMQ interface on csp_server_client on Ubuntu 22.04

danieldavidson, 2023-10-01:
On Ubuntu 22.04, building with `buildall.py --build-system cmake`, `CMakeLists.txt` doesn't set `CSP_HAVE_LIBZMQ` so `csp_server_client` isn't being built with a ZMQ interface, which means it can't be used with the `zmqproxy` that **is** being built. If I update `CMakeLists.txt` to set `CSP_HAVE_LIBZMQ=1`, I'm getting an error about an undefined reference to `csp_zmqhub_init`... but it's not clear to me why `csp_zmqhub_init` isn't available when I can see that we're making it into the `if(LIBZMQ_FOUND)` block of `src/interfaces/CMakeLists.txt`. Appreciate you helping me understand. 🙏 

```
[46/49] Linking C executable examples/csp_server_client
FAILED: examples/csp_server_client 
: && /usr/bin/cc -Os -DNDEBUG  examples/CMakeFiles/csp_server_client.dir/csp_server_client.c.o examples/CMakeFiles/csp_server_client.dir/csp_server_client_posix.c.o -o examples/csp_server_client  -Wl,-rpath,/src/build  libcsp.so && :
/usr/bin/ld: examples/CMakeFiles/csp_server_client.dir/csp_server_client.c.o: in function `main':
csp_server_client.c:(.text.startup+0x146): undefined reference to `csp_zmqhub_init'
collect2: error: ld returned 1 exit status
```

yashi, 2023-10-03:
Right, `examples/buildallpy` is, as you can see from the directory name,  an example script.

Try 
```
$ rm -rf build && cmake -S . -B build && cmake --build build
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/lib/ccache/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/lib/ccache/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- No build type selected, default to MinSizeRel
-- Looking for include file sys/socket.h
-- Looking for include file sys/socket.h - found
-- Looking for include file arpa/inet.h
-- Looking for include file arpa/inet.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.8.1") 
-- Checking for one of the modules 'libzmq'
-- Found /usr/lib/x86_64-linux-gnu/libzmq.so 4.3.4
-- Checking for one of the modules 'libsocketcan'
-- Found /usr/lib/x86_64-linux-gnu/libsocketcan.so 0.0.12
-- Configuring done (0.5s)
-- Generating done (0.0s)
-- Build files have been written to: /home/yashi/src/libcsp/build
[42/42] Linking C shared library libcsp.so
```

Or, read https://github.com/libcsp/libcsp/blob/develop/doc/INSTALL.md#using-cmake

yashi, 2023-10-03:
@danieldavidson, would you please test the PR?  I applogize deleting the mis-informing comment. I was wrong.

PR #464 (2023-10-04): cmake: Define CSP_HAVE_LIBZMQ and CSP_HAVE_LIBSOCKETCAN
CMake tries to find libzmq and libsocketcan and compile `interfaces/csp_if_zmqhub.c` and `drivers/can/can_socketcan.c` but it doesn't define `CSP_HAVE_LIBZMQ` and `CSP_HAVE_LIBSOCKETCAN` respectively even when `pkg-config` detects them.

When these defines are not defined, `csp_if_zmqhub.c` effectively becomes an empty file, and `can_socketcan.c` doesn't use any socketcan APIs.  Thus, the final libcsp.so isn't linked against the found libraries.

Define `CSP_HAVE_LIBZMQ` and `CSP_HAVE_LIBSOCKETCAN` in `csp_autoconfig.h` so that compile ti [... cut]
commit b70fa01a (2023-10-04): cmake: Define CSP_HAVE_LIBZMQ and CSP_HAVE_LIBSOCKETCAN

CMake tries to find libzmq and libsocketcan and compile
interfaces/csp_if_zmqhub.c and drivers/can/can_socketcan.c but it
doesn't define CSP_HAVE_LIBZMQ and CSP_HAVE_LIBSOCKETCAN respectively
even when pkg-config detects them.

When these defines are not defined, csp_if_zmqhub.c effectively
becomes an empty file, and can_socketcan.c doesn't use any socketcan
APIs.  Thus, the final libcsp.so isn't linked against the found
libraries.

Define CSP_HAVE_LIBZMQ and CSP_HAVE_LIBSOCKETCAN in csp_autoconfig.h
so that compile time condition works  [... cut]
closed 2023-10-04 by yashi