libcsp/libcsp #477: No reliable data transmission with RDP

alegnani, 2023-11-02:
Hi all,

I was modifying the `csp_server.c` and `csp_client.c` files to test whether transmitting data over a lossy connection with RDP actually is reliable. I implemented a proxy that drops 10% of the packets to see if this is the case.
Each packet contains an increasing counter to check for dropped packets. With the code files attached below I could not get a reliable connection to work. Additionally I had  to always create a new connection as there is, from what I could find, no way to tell if a connection is still alive. Same goes for checking if a message has actually been sent as the return type of `csp_send()` is void.

Am I missing something on how to properly get RDP to work?
Any guidance would be greatly appreciated.
Thank  you
[files.zip](https://github.com/libcsp/libcsp/files/13240975/files.zip)

johandc, 2023-11-04:
Your client spawns 5 new connections per second, one for each packet.

With RDP you are supposed to create one persistant connection and call csp_send() in a loop on the client, to continously stream data through.

Your server is correct in that it accepts a connection, and within this connection it starts a csp_read() loop.

Opening so many connections wont work because there is a conn close timeout of several seconds. So therefore they will pile up.

```c

/* Client task sending requests to server task */
void client(void) {

	csp_print("Client task started\n");

	unsigned int count = 0;

	while (1) {

		usleep(200000);

		/* Send data packet (string) to server */

		csp_conn_t * conn = csp_connect(CSP_PRIO_NORM, server_address, SERVER_PORT, 1000, CSP_O_RDP);
		if (conn == NULL) {
			/* Connect failed */
			csp_print("Connection failed\n");
			continue;
		}

		/* 2. Get packet buffer for message/data */
		csp_packet_t * packet = csp_buffer_get(100);
		if (packet == NULL) {
			/* Could not get buffer element */
			csp_print("Failed to get CSP buffer\n");
			return;
		}

		/* 3. Copy data to packet */
        char msg[512];
        memset(msg, '\0', 512);
        sprintf(msg, "HelloWorld:%d", count);
        int len = strlen(msg);
        printf("Len: %d\n", len);
        memcpy(packet->data, msg, len + 1);

		/* 4. Set packet length */
		packet->length = len + 1; /* include the 0 termination */

		/* 5. Send packet */
		// if (csp_rdp_send(conn, packet) != CSP_ERR_NONE) {
        //     csp_print("Could not send packet\n");
        // }
        csp_send(conn, packet);
        csp_print("Sent packet\n");

		/* 6. Close connection */
		csp_close(conn);
        count++;
	}

	return;
}
/* End of client task */
```

alegnani, 2023-11-04:
Having only one connection it works. Thank you
 As a follow-up I added packet corruption and enabled the CSP_O_CRC32 flag. Despite using CRC after a while the connection is dropped or packets arrive corrupted.
Is there any way to know, from the client side, if a certain connection is still open?
Any ideas why some packets still arrive corrupted? I am currently corrupting 1% of the bytes, which should be handled just fine by CRC32.

johandc, 2023-11-04:
What version of CSP are you using?

From 2.1 and onwards (current develop branch) the default behaviour is to protect both the header and the frame contents with the CRC. Prior to this, in version 1.6 only the payload is protected. And CSP was relying only on lower layer checksums.

alegnani, 2023-11-05:
I am using the latest commit of the development branch. I think I found out why it was not working:
Given that the corruption of packets would require more ACKing and retransmission of packets, and the timeout of `csp_read()` was quite short, the call would timeout. Then, given how the example code is structured, the connection would be closed. Simply increasing the timeout of `csp_read` fixes the problem.
One question remains: How can I know, when using RDP, if a connection has been closed by the other party? Is there any function aiding with that?

closed 2023-11-05 by alegnani