libcsp/libcsp #479: Build in debug mode with waf

AngelPerezM, 2023-11-08:
Hi all,

We are currently using `libcsp` for an university cube-sat project and it with FreeRTOS.
To test such integration, we would like to build it in a debug mode through the waf script (i.e.: -g flag for GCC).

The `python3 ./waf -h` command does not give any instructions to do it.

Kind regards,

Ángel Pérez Muñoz
Universidad Politécnica de Madrid (UPM)

yashi, 2023-11-09:
Unlike Meson or CMake, Waf doesn't provide built-in "Debug Build" feature, IIRC. Fortunately, if you don't specify `CFLAGS` by yourself, the default build option includes `-g`.  Check https://github.com/libcsp/libcsp/blob/develop/wscript#L76.

Hope that's what you want.

AngelPerezM, 2023-11-09:
Yes, that's what I was looking for.

Thanks for the quick response!
Ángel

closed 2023-11-09 by AngelPerezM