libcsp/libcsp #481: Compiling the library for Raspberry Pi Pico

evcegarra, 2023-11-09:
Hello,

I am interested in developing CSP inside freeRTOS on my raspberry pi pico. The main problem I encounter is that this board, does not have a way to access it via ssh or to its command terminal to compile the library into this structure.

The main information I found was that I needed to crosscompile the code using gcc-arm-none-eabi, but the toolchain.cmake file does not appear to be working or it is not provided. 

Do you have any recommendation on how to compile your library for ARM CortexM0+ MCUs? It will be appreciated. If you have the library already precompiled it will be good too.

Thank you in advance!

yashi, 2023-11-09:
Hi,

I don't use FreeRTOS but it might be better to use Meson to build libcsp.  CMake support was started to integrate libcsp with Zephyr RTOS.  And I personally build libcsp with CMake against Zephyr and Linux.

OTOH, we have a [sample repo](https://github.com/libcsp/libcsp-freertos) to build libcsp against FreeRTOS with Meson and this repos is used to test libcsp for every PR.  You can find build report under GitHub Actions,  such as https://github.com/libcsp/libcsp/actions/runs/6812949903/job/18526453876.

Hope this helps.

yashi, 2023-11-09:
BTW, Raspberry Pi Pico is also [supported by Zephyr RTOS](https://docs.zephyrproject.org/latest/boards/arm/rpi_pico/doc/index.html).  And [this](https://github.com/yashi/libcsp-zephyr) is how you use it.

evcegarra, 2023-11-10:
Hello,

Thank you for your fast response! I am using the raspberry pi pico just as a check to to develop a bit with CSP, because I will later implement it in my university CubeSat which will probably use an ARM MCU architecture, so that is why I am interested into compiling the library for this architecture to see how it works.

So I am interested in developing CSP with freeRTOS on an ARM MCU. 
My project structure looks like this:
[image]

So I am really interested into compiling the CSP library for ARM architectures and then linking the header files in my project's CMakeLists.txt file. Any feedback is welcomed!

yashi, 2023-11-10:
Did you try https://github.com/libcsp/libcsp-freertos ?

evcegarra, 2023-11-10:
Yes, I have tried. The main issue for me, is that the freeRTOS kernel in that repository is not the one specific for my project (that can be solved easily), but the problem comes with the meson.build file, this file builds for my computer architecture, and not for the ARM M0+ MCU, so  I will need either way a toolchain.txt file to tell meson compiler that instead of compiling for my computer architecture x86, it will need to do it using the gcc-arm-none-eabi compiler.
if this is sucessfull, it will generate the .u2f files to upload to my boards's MCU. I am missing this toolchain file to tell the compiler to build for another MCU architecture.

I wondering how other cubesats that use embbeded MCUs managed to make it.

yashi, 2023-11-10:
Ok.  In that case, you have two ways

### Port libcsp Meson to your architecture

As you already know, you have to provide a file, my-cortex-m0.txt, for cross compilation. And do

```
meson setup --cross-file my-cortex-m0.txt builddir
meson compile -C builddir
```
You can use the sample repo to start with.

Ref) https://mesonbuild.com/Cross-compilation.html

### Port libcsp CMake to FreeRTOS

- Make sure you can build any sample, i.e blinky, against FreeRTOS for your archtecture.
  - For RPi Pico:
    - https://www.google.com/search?q=build+freertos+for+raspberry+pi+pico
    - Many seem to be using [Pico SDK](https://github.com/raspberrypi/pico-sdk).
- Integrate libcsp into your build env. (But this should fail because it doesn't support FreeRTOS right now)
- Add FreeRTOS support to all of the libcsp's CMakeLists.txt files. (Search for "Linux" or "Zephyr"; that's where you need to add FreeRTOS support).

Hope this helps.

evcegarra, 2023-11-12:
I am trying to build the library using Meson as sugegsted. I have created a simple project to test it. This is the project structure:

`vladik@vladik:~/Documentos/pico/testing$ ls
a.out  builddir  cross_file.txt  libcsp  main.c  meson.build
` 

As you can see, I have made first a meson.build file for my project with the following content:
`project('my_project', 'c')
executable('my_program', 'main.c')
`
I have also created a cross_file.txt file to cross compile this main.c file into the RP2040 MCU and configured as follows:
`[binaries]
c = '/usr/bin/arm-none-eabi-gcc'
cpp = '/usr/bin/arm-none-eabi-g++'
ar = '/usr/bin/arm-none-eabi-ar'
ld = '/usr/bin/arm-none-eabi-ld'
strip = '/usr/bin/arm-none-eabi-strip'

[built-in options]
c_args = ['--specs=nosys.specs', '-mcpu=cortex-m0plus']
c_link_args = ['--specs=nosys.specs', '-mcpu=cortex-m0plus']

[host_machine]
system = 'bare-metal'
cpu_family = 'arm'
cpu = 'cortex-m0plus'  # e.g., 'cortex-m4'
endian = 'little'`

When running `meson build --cross-file cross_file.txt` it generates correctly the `a.out` file for my ARM architecture. This means, the cross compiler is working correctly. 

Now, when trying to cross compile the libcsp library using this same cross compiling file. I get the following error:

`vladik@vladik:~/Documentos/pico/testing/libcsp$ meson build --cross-file ../cross_file.txt
The Meson build system
Version: 0.61.2
Source dir: /home/vladik/Documentos/pico/testing/libcsp
Build dir: /home/vladik/Documentos/pico/testing/libcsp/build
Build type: cross build
Project name: csp
Project version: 2.0
C compiler for the host machine: /usr/bin/arm-none-eabi-gcc (gcc 10.3.1 "arm-none-eabi-gcc (15:10.3-2021.07-4) 10.3.1 20210621 (release)")
C linker for the host machine: /usr/bin/arm-none-eabi-gcc ld.bfd 2.38-3ubuntu1
C compiler for the build machine: cc (gcc 11.4.0 "cc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0")
C linker for the build machine: cc ld.bfd 2.38
Build machine cpu family: x86_64
Build machine cpu: x86_64
Host machine cpu family: arm
Host machine cpu: cortex-m0plus
Target machine cpu family: arm
Target machine cpu: cortex-m0plus
Found Pkg-config: NO
Found CMake: NO
Run-time dependency yaml-0.1 found: NO (tried pkgconfig and cmake)
Neither a subproject directory nor a yaml.wrap file was found.
Subproject  yaml is buildable: NO (disabling)
Library c found: YES

src/arch/meson.build:8:1: ERROR: **Problem encountered: invalid system bare-metal**

A full log can be found at /home/vladik/Documentos/pico/testing/libcsp/build/meson-logs/meson-log.txt
`

How can I modify the meson.build file inside the libcsp to avoid this error. Any ideas?

yashi, 2023-11-13:
You are not following how we build libcsp with FreeRTOS using Meson demonstrated in https://github.com/libcsp/libcsp-freertos.

Make sure to build, say, hello world for your RPi with FreeRTOS first.  Then, integrate libcsp.  Unlike Linux, we do NOT explicitly build `libcsp.a` then link to  your application.  Instead, libcsp will be built as a part of the build process.

evcegarra, 2023-11-13:
Thanks for the update! Really appreciate it.
I think the raspberry pi pico SDK only allows compilation using CMake. It does not have support to build with Meson (and I am noe xpert to configure it myself).
I got the grasp on how to configure freeRTOS + libcsp for other MCUs that support compilation with Meson (quite easy, just have to call the freeRTOS-kernel for your MCU architecture) and integrate libcsp.
From your previous response, I can tell that libcsp is not configured to build with freeRTOS using CMake. Will you update it in the future? Maybe all the CMake users can benefit.

yashi, 2024-01-10:
I'm closing this for now.  If you have more question / commnets, please reopen it.

closed 2024-01-10 by yashi