libcsp/libcsp #484: csp with UART

cspuart, 2023-11-13:
Hi, I'm trying to get csp running on UART on an RPi. I build it with meson ok (eventually by commenting out can_device  which it can not find/recognise). I have tested it with the information in examle.md for the ZMQPROXY and UART cases as described below:

For the ZMQPROXy(success):
I can run the example ok with the ZMQPROXY and it shows the Hello World messages.

For the UART case (fail):
In my test setup, I have two UARTS at /dev/ttyUSB0 and /dev/ttyUSB1 which are connected back to back with a crossover cable e.g. Tx0 to Rx1 and Rx0 to Tx1. I invoke as follows : ./csp_client -k /dev/ttyUSB0 and ./csp_server -k /dev/ttyUSB1 respectively.  There is no observable output on the console. The only think I notice is that the Tx0 line on the UART associated with the client is flashing (as is the Rx1 line).

Can anyone give any pointers to what I'm doing incorrectly? I feel like I'm missing a piece of the jigsaw in my understanding.
I feel like I should be specifying addresses for the client and server but it doesn't seem to make a difference when I do. 
I have tried the following ./csp_client -k /dev/ttyUSB0 -a 3 -C 2 and csp_server  -k /dev/ttyUSB1 -a 2 without success.
Also do I need to specify a routing table and how? 

What I want to be able to do is to setup the client and server such that I can specify an address and a port for each message being sent by the client and the server to only respond to specific incoming messages.
Apologies for the long message.

Yours 
cspuart

yashi, 2023-11-13:
I'm assumeing you are using Linux on RPi.

Have you check your serial connection between `ttyUSB0` and `ttyUSB1` with some serial comunication app, such as `picocom`, `minicom`, or just `cat` with preconfigured port with`stty`?

> Also do I need to specify a routing table and how?

Yes.  Just like `csp_server_client.c` does. See https://github.com/libcsp/libcsp/blob/develop/examples/csp_server_posix.c

cspuart, 2023-11-13:
Thanks for your reply and the pointers. Yes, its Linux on RPi. I have tested the serial port connection with picocom and it all works fine. What I transmit on one port gets received on the other port and vice versa. 

I think I am missing some understanding of what I need to do to get the client and server code samples running over the serial ports. Should they work together as is (out-of-the-box... so to speak) by just specifying the appropriate kiss port as I have outlined in my previous message. Or are these pieces of code incomplete and need further initialization/set up.

Would you mind elaborating on routing table link you included if you have time.

Thanks for your help. Its much appreciated. I have some experience with Serial comms and Network comms but csp is very new to me.

yashi, 2023-11-13:
Ah, that is #459.

For the time being, would you please apply the following patch?

```diff
diff --git a/examples/csp_client.c b/examples/csp_client.c
index d0ba0ae7..14363e65 100644
--- a/examples/csp_client.c
+++ b/examples/csp_client.c
@@ -205,6 +205,7 @@ int main(int argc, char * argv[]) {
             csp_print("failed to add KISS interface [%s], error: %d\n", kiss_device, error);
             exit(1);
         }
+        default_iface->addr = client_address;
         default_iface->is_default = 1;
     }
 #if (CSP_HAVE_LIBSOCKETCAN)
diff --git a/examples/csp_server.c b/examples/csp_server.c
index 4810c89e..756f07a1 100644
--- a/examples/csp_server.c
+++ b/examples/csp_server.c
@@ -191,6 +191,7 @@ int main(int argc, char * argv[]) {
             csp_print("failed to add KISS interface [%s], error: %d\n", kiss_device, error);
             exit(1);
         }
+        default_iface->addr = server_address;
         default_iface->is_default = 1;
     }
 #if (CSP_HAVE_LIBSOCKETCAN)
```

cspuart, 2023-11-14:
I don't have access to RPi today. So instead I modified the code and tested it on a VMware Ubuntu Image. 
I had to use two different USB to UART cables also. So I had to ensure the new cables were ok to run within the Ubuntu image. I tested them back to back with picocom. The result was that all was fine with picocom , the USB to UART cables with the Ubuntu Image.
However, when I run the csp_client and csp_server  as per below, I get a VMware crash. It may be just the VMware image so I will test it later with the RPi.  
./csp_server -k /dev/ttyUSB1 -a 2
./csp_client -k  /dev/ttyUSB0 -a 3 -C 2

The crash occurs as soon as  I connect the Transmit line of the client to the Receive line of the server.

Again thanks for your help.

yashi, 2023-11-19:
Post a backtrace with `gdb`.

yashi, 2024-01-10:
I'm closing this for now.  If you have more question / commnets, please reopen it.

closed 2024-01-10 by yashi