libcsp/libcsp #491: RDP and CRC32 dropping packets

alegnani, 2023-11-16:
As the description says, when using both RDP and CRC32, packets get dropped.
I tried using both RDP and CRC32 separately and they work fine by themselves.

The issue is encountered when sending packets, with a monotonically increasing counter as a message, over a connection that randomly corrupts bytes.
Could it be that the ACK gets send when receiving the packet before it gets dropped due to a wrong CRC?

yashi, 2023-11-20:
Care to share how to reproduce?

alegnani, 2023-11-20:
Sorry forgot to actually upload the code.
`csp_server.c` and `csp_client.c` are replacements for the files in the example folder and can be built with `python buildall.py`.
To run them I used: `./build/examples/csp_server -z localhost -a 2` and `./build/examples/csp_client -z localhost -a 1 -C 2`.
The proxy that introduces corruption can be run with `python lossy_zmq_proxy.py`.

I am observing that packets arrive duplicated at the server.

[code.zip](https://github.com/libcsp/libcsp/files/13411773/code.zip)

yashi, 2023-11-20:
I can't even make it work with `CSP_O_NONE`, can you?

alegnani, 2023-11-20:
I uploaded the wrong file. It should be `CSP_O_RDP | CSP_O_CRC32`.

These are the tests I performed:
-  `CSP_O_RDP`, packet_loss = 0.1 and no corruption:
   - Sending increasing counter works
- `CSP_O_CRC32`, no packet_loss and corruption = 0.02:
   - Sending the same message over and over does not arrive corrupted.
   - If the CRC32 is wrong it gets dropped
- `CSP_O_CRC32 | CSP_O_RDP`, packet_loss = 0.1 and corruption = 0.02:
   - Sending increasing counter does not work
   - Sometimes a duplicated message arrives
   - Sometime a message gets skipped
   - Same thing happens with packet_loss = 0.0

yashi, 2023-11-21:
Is this due to ZMQ?

> A successful invocation of zmq_send() does not indicate that the message has been transmitted to the network, only that it has been queued on the socket and ØMQ has assumed responsibility for the message.

> When a ZMQ_PUB socket enters the mute state due to having reached the high water mark for a subscriber, then any messages that would be sent to the subscriber in question shall instead be dropped until the mute state ends.

yashi, 2023-11-21:
```c
	for (;;) {
		csp_packet_t * packet;
		while ((packet = csp_read(conn, 100)) != NULL) {
			if (csp_conn_dport(conn) == SERVER_PORT) {
				/* Process packet here */
				int recv;
				sscanf((char *)packet->data, "HelloWorld:%d", &recv);
				csp_print("Packet received on SERVER_PORT: %d\n", recv);
				csp_buffer_free(packet);
				if (recv != server_received) {
					csp_print("expected:%d, got: %d\n", server_received, recv);
					return;
```
Once it's out of sync, you will get OOM.

alegnani, 2023-11-21:
Thank you very much for investigating the issue.
If this is only an issue with ZMQ then it's not an issue for me.
This program was just to sanity check whether the communication is actually reliable or not, before using it on our cubesat.
There should be no problems when using it with CAN right? (We will definitively test that later on.)
If the point above is the case this issue can be marked as resolved.

moonlight83340, 2024-05-01:
> Thank you very much for investigating the issue. If this is only an issue with ZMQ then it's not an issue for me. This program was just to sanity check whether the communication is actually reliable [...]

Hi,
Sorry, it's been a while, but I'd like to know if you defined CSP_21?
https://github.com/libcsp/libcsp/blob/f33f496ca3635e59103599df0a95a0dcee9c201c/src/csp_crc32.c#L95
The CRC will consider both the message header and the data; otherwise, the CRC would only be effective on the data part and not on the header.

When I attempt to introduce corruption from the beginning of the data segment (without corrupting the message header), I don't encounter any issues with RDP + CRC. (CSP_21 is not define)

moonlight83340, 2024-07-10:
Hi,

It appears the issue occurred because the server could potentially accept a corrupted message if the CRC32 flag from the client was also corrupted. However, we can configure the server to only accept messages with CRC32.

You can review my issue and the response here: https://github.com/libcsp/libcsp/issues/596

To implement this, you can add the following line:
`sock.opts = CSP_SO_CRC32REQ;`
Here : https://github.com/libcsp/libcsp/blob/develop/examples/csp_server.c#L43, and then run the example.

This setting will add a security check on the server side. I wasn't aware that we could use this option on the server side before.

I would also recommend defining CSP_21 so that CRC32 applies to the entire packet (header + data).

closed 2024-08-06 by alegnani