libcsp/libcsp #494: Zephyr server-client example hangs up

ignacio-gr, 2023-11-22:
My problem is that when I try to get the server-client zephyr example going on a microcontroller (nucleo l496zg on my case) , it does it well twice and then the client hangs up on a csp_buffer_get call.
That is when I try to do it with the loopback, when I try to use can to comunicate two devices, it doesnt send anything.

Any clue why this happens?

ignacio-gr, 2023-11-28:
In order to compile for nucleo_l496zg you need to add an overlay inside of contrib/zephyr/samples/server-client/boards which needs to be called nucleo_l496zg.overlay and has this content:
```c
#include <freq.h>

&can1 {
    status = "okay";
    pinctrl-0 = <&can1_tx_pd1 &can1_rx_pd0>;
    pinctrl-names = "default";
    bus-speed = < DT_FREQ_K(250) >;
};
``` 
Then you need to modify
# https://github.com/libcsp/libcsp/blob/98103db3bb16ca79bf209081748c2c72a9b61ad8/contrib/zephyr/samples/server-client/main.c#L183 
by
```c
		const struct device * device = DEVICE_DT_GET(DT_NODELABEL(can1));
``` 
And enabling can tx by setting `CONFIG_CSP_HAVE_CAN=y` on prj.conf.

Then you just need to execute the application and flash it onto the microcontroller and anything will appear on the CAN interface.
```shell
west build -p always -b nucleo_l496zg contrib/zephyr/samples/server-client
west flash
```

yashi, 2023-11-29:
I don't know what's going on but the current tip of `develop` branch works for us.  We need more info:
- what and how exactly does it hangs up? 
- How did you setup your system (do you have two nucleo_l496zg, or you are just testing loopback?)

Anyway, please help us to help you with more info.

Thanks

ignacio-gr, 2023-11-29:
I solved the CAN problem, it was just a mistaken modification.

For the buffer problem I was testing it with loopback. It works fine 3 times and then csp_buffer_get returns an error (-115 i believe).

yashi, 2023-11-29:
> I solved the CAN problem, it was just a mistaken modification.

Good to hear.

> For the buffer problem I was testing it with loopback. It works fine 3 times and then csp_buffer_get returns an error (-115 i believe).

Let me clarify.

1. You said you solved CAN problem.  Does it mean two device can talk each other via CSP?
2. The buffer problem you just mentioned is with loopback.
3. So, loop back doesn't work but two device can talk each other.  I that what you are saying?

ignacio-gr, 2023-11-29:
The problem I just mention happens both on the loopback and can. I was able to send and receive three packets, but that happens on the fourth iteration of the client loop.

yashi, 2023-11-29:
OK.  Do you have any debug log?

ignacio-gr, 2023-11-29:
Nothing shows up on the debug log, I tried to trace the error and got to this point.
https://github.com/libcsp/libcsp/blob/98103db3bb16ca79bf209081748c2c72a9b61ad8/src/arch/zephyr/csp_queue.c#L54
That call is what returns the error

yashi, 2023-11-30:
That means you don't have enough memory, but that shouldn't be the case since we should have initialized to `CSP_BUFFER_COUNT` in `csp_buffer_init()`.

BTW, `-115` in Zephyr is `ENETDOWN`.  Are you sure you get `-115`? That's more like an error from `csp_send()`.

yashi, 2023-11-30:
> Nothing shows up on the debug log

Right.  Zephyr port doesn't inegrate logging subsystem with `csp_print()` yet.  That's something we need to design how to do it. (many satellite has it's own way to collect components logs and I haven't come up with a good one yet).

ignacio-gr, 2023-12-04:
> That means you don't have enough memory, but that shouldn't be the case since we should have initialized to `CSP_BUFFER_COUNT` in `csp_buffer_init()`.

Shouldn't there be a Kconfig option to set CSP_BUFFER_COUNT?

> BTW, `-115` in Zephyr is `ENETDOWN`. Are you sure you get `-115`? That's more like an error from `csp_send()`.

Thats true, Im not really sure it was -115, I looked it up when I got it and it was something about the memory

yashi, 2023-12-06:
What do you see when you run the following code?

```c
#include <stdio.h>
#include <csp/csp.h>

#define TEST_COUNT (10)

static void do_test(void)
{
	csp_packet_t *bufs[CSP_BUFFER_COUNT];
	int i = 0;

	while (true) {
		bufs[i] = csp_buffer_get(1);
		if (bufs[i] == NULL) {
			printf("Got NULL\n");
			break;
		}
		else {
			printf("%02d: p == %p\n", i, bufs[i]);
		}

		i++;

	}

	if (i != CSP_BUFFER_COUNT) {
		printf("Failed to allocate %d buffers\n", CSP_BUFFER_COUNT);
	}

	for (i = 0; i < CSP_BUFFER_COUNT; i++) {
		csp_buffer_free(bufs[i]);
		printf("%02d: free %p\n", i, bufs[i]);
	}
}

int main(void)
{
	int i;

	csp_init();

	printf("Hello World! %s\n", CONFIG_BOARD);

	for (i = 0; i < TEST_COUNT; i++) {
		do_test();
		printf("done %d\n", i);
	}

	return 0;
}
```

yashi, 2023-12-06:
> Shouldn't there be a Kconfig option to set CSP_BUFFER_COUNT?

No.  It's not Kconfig variable yet. It should be and I'd love to take a PR for it.  But it's still defined in the top level `CMakeLists.txt` and you can find it in 
`build/modules/libcsp/build/include/csp/autoconfig.h`

ignacio-gr, 2023-12-11:
> What do you see when you run the following code?

I get the following:
```
*** Booting Zephyr OS build zephyr-v3.5.0-33-g11b7c988eb8c ***
Hello World! nucleo_l496zg
00: p == 0x20002470
01: p == 0x200025a4
02: p == 0x200026d8
03: p == 0x2000280c
04: p == 0x20002940
05: p == 0x20002a74
06: p == 0x20002ba8
07: p == 0x20002cdc
08: p == 0x20002e10
09: p == 0x20002f44
10: p == 0x20003078
11: p == 0x200031ac
12: p == 0x200032e0
13: p == 0x20003414
14: p == 0x20003548
15: p == 0x2000367c
16: p == 0x200037b0
17: p == 0x200038e4
18: p == 0x20003a18
19: p == 0x20003b4c
20: p == 0x20003c80
21: p == 0x20003db4
22: p == 0x20003ee8
23: p == 0x2000401c
24: p == 0x20004150
25: p == 0x20004284
26: p == 0x200043b8
27: p == 0x200044ec
28: p == 0x20004620
29: p == 0x20004754
30: p == 0x20004888
```

ignacio-gr, 2023-12-11:
Maybe we cant use posix msg queue APIs without using a posix board?

yashi, 2023-12-11:
> I get the following:

Did you compile and run the exact same code?

Where did `printf("Hello World! %s\n", CONFIG_BOARD);` go?

> ```
> *** Booting Zephyr OS build zephyr-v3.5.0-33-g11b7c988eb8c ***
> 01: p == 0x200025a4 <--- where is `00`?
> 02: p == 0x200026d8
> 03: p == 0x2000280c
> 04: p == 0x20002940
> 05: p == 0x20002a74
> 06: p == 0x20002ba8
> 07: p == 0x20002cdc
> 08: p == 0x20002e10
> 09: p == 0x20002f44
> 10: p == 0x20003078
> 11: p == 0x200031ac
> 12: p == 0x200032e0
> 13: p == 0x20003414
> 14: p == 0x20003548
> Got NULL
> 00: free 0x20002470  <---- How did you get this?
> 01: free 0x200025a4
> 02: free 0x200026d8
> 03: free 0x2000280c
> 04: free 0x20002940
> 05: free 0x20002a74
> 06: free 0x20002ba8
> 07: free 0x20002cdc
> 08: free 0x20002e10
> 09: free 0x20002f44
> 10: free 0x20003078
> 11: free 0x200031ac
> 12: free 0x200032e0
> 13: free 0x20003414
> 14: free 0x20003548
> ```

yashi, 2023-12-11:
> Maybe we cant use posix msg queue APIs without using a posix board?

Do you mind elaborating a bit?

It appears you are using the Nucleo L496ZG board. Can you confirm this? Knowing the specific hardware can help tailor the advice to your setup.
Also you mentioned POSIX message queue APIs. Could you provide more details on why you are considering them and what specific functionality you are trying to implement?  As you can see from `arch/zephyr/csp_queue.c`, we are using [`k_msgq`](https://docs.zephyrproject.org/latest/kernel/services/data_passing/message_queues.html)

ignacio-gr, 2023-12-11:
I solved the issue.
It was a commit I did long time ago when trying to fix these warnings.
[image]

closed 2023-12-11 by yashi