libcsp/libcsp #499: Storing the date and time in the code make a build non reproducible

ElouanPetereau, 2024-01-09:
Hello,

I observed that that, when buildign the library, the compile date and time are stored in the generated binary.  Due to this, I cannot assure that a given version of a software version using your library is reproducible since this is the only difference observable in my binary.    
For information, this happen in the `do_cmp_ident()` function of the `src/csp_service_handler.c` file.

Do you plan to do something to allow full reproducibility of the project?    
Ideas could be to add a build flag to give a defined date, use the GIT sha instead (and an hardcoded number if built from outside a GIT repository), add a real version number tracking.

For now I just patched it with an hardcoded value but I am probably not the only person interested in having a reproducible build.

Thanks!

yashi, 2024-01-10:
ah, that's very true.  Thank you for pointing out.  I think a first step would be to make them build option.  Or if @johandc doesn't object we can remove them.

These are the candidates for removal.

- date
- time
- hostname

FYI: https://reproducible-builds.org/

yashi, 2024-01-29:
@ElouanPetereau Would you mind tesing the PR?

ElouanPetereau, 2024-01-29:
> @ElouanPetereau Would you mind tesing the PR?

Thank you I tested the `CMAKE` version with different build time and users and it gave me completely reproducible builds.

I also took the liberty to make some comment on your PR (https://github.com/libcsp/libcsp/pull/506#pullrequestreview-1848628055)

PR #506 (2024-02-01): Add reproducible builds
This PR adds support for [reproducible builds](https://reproducible-builds.org/) to libcsp.

This fixes #499.
closed 2024-02-01 by yashi