libcsp/libcsp #500: Setting interface address

Robert-mBryonics, 2024-01-10:
I cannot get the -a argument to change the interface address when running the commands ./build/examples/csp_server -k /dev/ttyUSB0 -a 2 and ./build/examples/csp_client -k /dev/serial0 -a 1 -C 2. See below for terminal output. 

Initialising CSP
Connection table
[00 0x562cbc97ca60] S:0, 0 -> 0, 0 -> 0 (17) fl 0
[01 0x562cbc97cb70] S:0, 0 -> 0, 0 -> 0 (18) fl 0
[02 0x562cbc97cc80] S:0, 0 -> 0, 0 -> 0 (19) fl 0
[03 0x562cbc97cd90] S:0, 0 -> 0, 0 -> 0 (20) fl 0
[04 0x562cbc97cea0] S:0, 0 -> 0, 0 -> 0 (21) fl 0
[05 0x562cbc97cfb0] S:0, 0 -> 0, 0 -> 0 (22) fl 0
[06 0x562cbc97d0c0] S:0, 0 -> 0, 0 -> 0 (23) fl 0
[07 0x562cbc97d1d0] S:0, 0 -> 0, 0 -> 0 (24) fl 0
Interfaces
LOOP       addr: 0 netmask: 14 dfl: 0
           tx: 00000 rx: 00000 txe: 00000 rxe: 00000
           drop: 00000 autherr: 00000 frame: 00000
           txb: 0 (0B) rxb: 0 (0B) 

KISS       addr: 0 netmask: 0 dfl: 1
           tx: 00000 rx: 00000 txe: 00000 rxe: 00000
           drop: 00000 autherr: 00000 frame: 00000
           txb: 0 (0B) rxb: 0 (0B) 

Server task started

Server - Ubuntu 22.04 on laptop 
Client - Raspberry Pi - Serial Tx Rx pins. 

Perhaps I'm missing something but should the -a flag not set the addresses? I've also tried adding a routing table entry directly in the both the Server and Client scripts. 

if (default_iface) {
    csp_rtable_set(1, -1, default_iface, CSP_NO_VIA_ADDRESS);
}

Any help would be much appreciated.

yashi, 2024-01-11:
apologize for inconvenient.

Please see https://github.com/libcsp/libcsp/issues/484#issuecomment-1809314790

Robert-mBryonics, 2024-01-11:
This worked a treat. Thanks a million.

closed 2024-01-11 by Robert-mBryonics