libcsp/libcsp #504: Documentation on how to develop new drivers

olmanqj, 2024-01-23:
Hi, I need some help developing the driver for the UART or CAN interface for my ebedded board. 

As the documentations states: "If you do not find your driver supported, it is quite simple to add a driver that conforms to the CSP interface." However I can't find which is this interface. Provided drivers are quite different between each others.

It would be nice to see a small tutorial or a generic driver. For example, what are the expected inputs and outputs for a driver.

Many thanks in advance!

yashi, 2024-01-24:
We've [merged another CAN layer](https://github.com/libcsp/libcsp/pull/426) last Nov.  This should help understand how you do it.

libcsp's "driver layer" is not really a device driver layer in Linux kernel or some other RTOS's sense.  It's more like a glue layer between our interface layer and underlying layer, which can be API given by your OS, middleware, device driver, or whatever you have on your board.

Let us know when you have any question.

yashi, 2024-01-28:
I'm closing this.
Please reopne it if you have more questions, or open a new one if you have a specific question.

closed 2024-01-28 by yashi