libcsp/libcsp #513: ZMQ with Python bindings client/server example

clythersHackers, 2024-02-13:
Hi I'm having trouble with getting the Python bindings client/server example to work over ZMQ

Python bindings in general are working: this works fine: 
LD_LIBRARY_PATH=build PYTHONPATH=build python3 examples/csp_server_client.py

So with ZMQ, I start the proxy like this:
zmqproxy -d

Then the server like this:
LD_LIBRARY_PATH=build PYTHONPATH=build python3 examples/python_bindings_example_server.py
Then the client like this:
LD_LIBRARY_PATH=build PYTHONPATH=build python3 examples/python_bindings_example_client.py -z localhost

The client returns error -3 from cmp_ident, the first call (timeout), I tried increasing timeout from 1s to 5s, the same.

I checked the server addresses used on client/server both = default 27

I suspected some problem with the proxy (ran, attached with gdb) it never receives anything

On the basis of some reports here, I redefined the ZMQ ports 6000->6111, 7000->7111, rebuilt & reinstalled, no luck

So then I tried inserting some garbage packets with telnet on port 6111
Then the proxy does respond, see below:

$ zmqproxy -d
Subscriber task listening on tcp://0.0.0.0:6111
Publisher task listening on tcp://0.0.0.0:7111
Capture/logging task listening on tcp://0.0.0.0:6111
Packet: Src 834, Dst 10099, Dport 38, Sport 17, Pri 1, Flags 0x26, Size 3
Packet: Src 665, Dst 9741, Dport 55, Sport 13, Pri 1, Flags 0x24, Size 3

Have these examples been tried recently? Am I alone in having this trouble?

TX!

clythersHackers, 2024-02-14:
I also tried the C examples csp_server and csp_client with -z localhost -t options; no evidence of zmqproxy getting any messages at all, it doesn´t even seem to wake up.

clythersHackers, 2024-02-14:
Solution found: zmqproxy.c:

const char * sub_str = "tcp://*:6111";
const char * pub_str = "tcp://0.0.0.0:7111";

I am not 100% sure of the reason, but I think it is because localhost on a  machine with IPV6, does not match 0.0.0.0

closed 2024-02-14 by clythersHackers

yashi, 2024-02-15:
Hi @clythersHackers,

Sorry not to respond to you in time.

Does it work if you give an IPv4 address?

clythersHackers, 2024-02-15:
> 
> Does it work if you give an IPv4 address?

OK; csp_client, csp_server are working now (the .c examples) but not yet the Python ones.

Running zmqproxy (hub) with my modification, it shows & logs CSP traffic, consistent with what I expected
I can also inject my own messages, from my own app and subscribe to them.

But this never shows up as generating any traffic on the hub at all;
LD_LIBRARY_PATH=build PYTHONPATH=build python3 examples/python_bindings_example_client.py -z 127.0.0.1 
(also trying with various combinations of -s and -a options to give addresses)

It always fails at the first cmp_ident call (no timeout) with no evidence of any message on the hub

I think I should see the CMP ident message followed by a response by the server (when it is running)

moonlight83340, 2024-05-08:
> > Does it work if you give an IPv4 address?
> 
> OK; csp_client, csp_server are working now (the .c examples) but not yet the Python ones.
> 
> Running zmqproxy (hub) with my modification, it shows & logs CSP traffic, consistent with what I expected I can also inject my own messages, from my own app and subscribe to them.
> 
> But this never shows up as generating any traffic on the hub at all; LD_LIBRARY_PATH=build PYTHONPATH=build python3 examples/python_bindings_example_client.py -z 127.0.0.1 (also trying with various combinations of -s and -a options to give addresses)
> 
> It always fails at the first cmp_ident call (timeout) with no evidence of **any** messages on the hub neither CMP request nor obviously any reply
> 
> I think I should see the CMP ident message followed by a response by the server (when it is running)
> 
> I can also run the server in python debugger, no evidence of any message received. I think the issue is ZMQ-related not CSP. I'm not that familiar with ZMQ, but it could help if there were more documentation of how it's supposed to be working, since ZMQ has many different ways of working, I think it's the hub/spoke model.

I recently tried testing the Python example, and encountered the same problem. 

I attempted it with both KISS and ZMQ, and received a timeout error for both. 
I also tried a simple transaction without cmp ident, and I'm not seeing a message being received by the proxy. 

Because the error occurred with KISS as well, I don't think it's a ZMQ-related issue.