libcsp/libcsp #514: bug in csp_crc32_verify

clythersHackers, 2024-02-15:
For what looks like historical reasons, this function tries to run the CRC32 twice;

Once  "with headers" and again "without headers"  (hmm)

(I notice a comment  that CSP_21 will default to "with headers" (makes sense))

However, in the first try, with headers, it does this, taking frame start & whole frame length

	crc = csp_crc32_memory(packet->frame_begin, packet->frame_length);

But of course that's not right, it should be:

	crc = csp_crc32_memory(packet->frame_begin, packet->frame_length-sizeof(crc));

Later on it uses memcmp to compare the last 4 bytes with the CRC, but you included it in computing the CRC.

(sometimes, eg with CCSDS CRC16,  a CRC including CRC itself comes out with zero, but not with this one (I think CRC32-C))

Shall I push the fix, it's trivial?

yashi, 2024-02-24:
As noted in c16ce5c8ecfe14ffb7f860b4bfebd078c2153ba2, we are doing it intentionally.

> crc32: now checks data+header too
> 
> In the transition period we do the following:
> 2.0 starts accepting packets with data+header checksum, but still supports data only.
> 2.1 starts sending packets with data+header checksum
> 2.2 stops accepting packets with data only checksum

yashi, 2024-03-20:
@clythersHackers I see the problem.  I'll create a PR but it might take a while (too busy).  Hope someone beat me to it

clythersHackers, 2024-03-20:
No stress, don't worry, if you see the original report, I posted the solution, "it should be"
I would push it myself, but as a newbie here, I wanted to check whether that is considered good behaviour first.

yashi, 2024-03-22:
Oops.  I've told my colleague to create a PR without realizing your offer.  Apologize.  Would you mind if we add you in the sign-off ?  If you are ok, give me your sign-off line.

clythersHackers, 2024-03-22:
I'm new here! Sorry, I have no idea what you mean by sign-off line.
I can confirm I've tested it, and I guess now it can be closed.

moonlight83340, 2024-03-22:
> I'm new here! Sorry, I have no idea what you mean by sign-off line. I can confirm I've tested it, and I guess now it can be closed.

No problem, the sign-off line as this format :

`Signed-off-by: Your Name <your.email@example.com>`

For the name, you should use your legal name instead of **Your name** .
And for the email, You should use the same email address you use in github.
(In fact it would be _with the same email address you are using to author the commit_ but I think this time it's ok)

PR #546 (2024-03-26): csp_crc32_verify: Exclude CRC size from packet length calculation
When verifying the CRC in csp_crc32_verify(), it is incorrect to calculate the CRC for the entire packet, including the CRC itself. 
In csp_crc32_append(), we generate a CRC value for the entire packet and then append this value at the end.
This commit corrects the issue by subtracting the size of the CRC from the frame_length, thus fixing the end address calculation.
This fix aligns with the proposal made by clythersHackers in https://github.com/libcsp/libcsp/issues/514.
Fixes #514
closed 2024-03-22 by clythersHackers