libcsp/libcsp #524: cmake build with freertos

JoseMorenoJ, 2024-02-25:
There is no `CMakeLists.txt` file in `libcsp\src\arch\freertos` so it complains when setting the CMAKE_SYSTEM_NAME variable to FreeRTOS. Is there a particular reason for this? I saw that the example in [libcsp-frertos](https://github.com/libcsp/libcsp-freertos) is using meson to build it.

yashi, 2024-02-25:
Hi @JoseMorenoJ 

Initially CMake support for libcsp is added sololy for Zephyr RTOS.  Since then, people started using it on Linux and others.  But I do not use FreeRTOS and don't have any target except libcsp-freertos, which is targetting POSIX system.

If you know how to setup FreeRTOS with libcsp using CMake please let us know.  I might be able to find time slot to add support for FreeRTOS and create test case for it.

yashi, 2024-02-28:
Sure! Ask me anything.

BTW, you can tell me how you build FreeRTOS. I might be able to figure out from there.

yashi, 2024-02-28:
I just added CMakeLists.txt to https://github.com/libcsp/libcsp-freertos. Instructions to build it are in the README.md, but I'm putting them here for your convenience:

```
git clone https://github.com/libcsp/libcsp-freertos
cd libcsp-freertos
git clone https://github.com/FreeRTOS/FreeRTOS-Kernel freertos
mkdir subprojects
git clone https://github.com/libcsp/libcsp subprojects/libcsp
cmake -B builddir -GNinja
ninja -C builddir
```

yashi, 2024-03-20:
@JoseMorenoJ Reopen this if you have any other question.

closed 2024-03-20 by yashi

adomasbaliuka, 2024-09-23:
@yashi while looking at the `libcsp` repository, I just noticed the same issue as JoseMorenoJ initially did: The CMake setup (adding a subdirectory that does not contain a `CMakeLists.txt`) is incorrect.

Would it be possible to remove the `add_subdirectory` [here](https://github.com/libcsp/libcsp/blob/cc71d1ad4156d63e6fee1ce2a10825fca394268c/src/arch/CMakeLists.txt#L6) and replace it with a useful error message?

I looked at your repo https://github.com/libcsp/libcsp-freertos  and it builds but I'm not yet sure how this relates to the files at `src/arch/freertos` not being part of the CMake build system...

yashi, 2024-09-24:
I really want someone to submit a PR to build libcsp against FreeRTOS on GitHub Actions or at least tell me how to build and run it on QEMU.  We have libcsp-freertos right now but it's targetting POSIX system.  If we can build and test libcsp + FreeRTOS on QEMU (`qemu-system-arm -machine mps2-an385`), we can add GitHub Actions to verify it.

adomasbaliuka, 2024-09-24:
Unfortunately it seems that I'll be stuck with using version v1.6, which has no CMake support at all. If that's the case, I won't be able to justify spending time on this, at least not this year. Sorry to bother you and thanks for the quick responses!

(I still think putting an error message in that place saying "CMake build for FreeRTOS is not supported yet" would be good in the meantime)

yashi, 2024-09-24:
@adomasbaliuka Thank you for your feedback, though.  Hope you like the PR.

As you might already know `libcsp-freertos` is not actually targetting freertos.  We need to fix it, too.

Let us know if you need any help :smiley: