libcsp/libcsp #531: Why clear timestamp_rx in csp_id_strip()?

yashi, 2024-02-28:
https://github.com/libcsp/libcsp/commit/7b6190b81c1c9741a3c1f825923ebb3320d417d4

Why do we need to set `timestamp_rx` to `0`?  The commit log says

> We need this to be cleared to avoid using a timestamp from an older packet

But as far as I can see, nobody is using `timestamp_rx`.

This causes unioned member `last_used` to be cleared to `0`.

johandc, 2024-03-01:
We use timestamp_rx somewhere, and it will be used for the GPS timesync feature that we are also working on. last_used, what is that used for? Isnt it extremely similar in idea, to put a timestamp when altered. But last_used could also be for stale buffer detection or similar.

yashi, 2024-03-01:
`last_used` is used in CAN.  Because `timestamp_rx` is sharing storage area with it, it gets `0.  

https://github.com/libcsp/libcsp/blob/develop/src/interfaces/csp_if_can_pbuf.c#L68-L78

```c
void csp_can_pbuf_cleanup(csp_can_interface_data_t * ifdata, int * task_woken) {

	uint32_t now = (task_woken) ? csp_get_ms_isr() : csp_get_ms();

	csp_packet_t * packet = ifdata->pbufs;
	csp_packet_t * prev = NULL;

	while (packet) {

		/* Perform cleanup in used pbufs */
		if (now - packet->last_used > PBUF_TIMEOUT_MS) {
```

This contion almost always gets `true`.

yashi, 2024-03-01:
What was the motivation to use union on this `struct`s in `struct csp_packet_s`?  To save some memory?

yashi, 2024-10-16:
@johandc Let me remove `timestamp_rx = 0` from `csp_id_strip()` for now. This code doesn't and shouldn't belong to CSP ID and nobody in the repo is using it right now, but causing harm to other part of the library.  Let's discuss in another issue or PR if you need `timestamp_rx = 0` for your GPS timesync.

closed 2024-10-31 by yashi